project-chip/connectedhomeip #14325: Darwin test failed twice, looks like commissioning error

woody-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4941174256?check_suite_focus=true
```
2022-01-25 19:11:43.460 ERROR   19:11:43.304 - PAIR OUT  : [1643137903304] [54650:145663] CHIP: [-] ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1293: CHIP Error 0x00000003: Incorrect state at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
1307
2022-01-25 19:11:43.460 ERROR   19:11:43.304 - PAIR OUT  : [1643137903304] [54650:145663] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1293: CHIP Error 0x00000003: Incorrect state
1308
```
 #### Proposed Solution
Fix

woody-apple, 2022-01-26:
Dup here: https://github.com/project-chip/connectedhomeip/runs/4931722557?check_suite_focus=true

woody-apple, 2022-01-26:
Another: https://github.com/project-chip/connectedhomeip/runs/4925758385?check_suite_focus=true

bzbarsky-apple, 2022-01-27:
So for that first run, we have:
```
2022-01-25T19:11:43.5041610Z 2022-01-25 19:11:43.456 ERROR   19:11:43.303 - PAIR OUT  : [0;32m[1643137903303] [54650:145667] CHIP: [CTL] Add NOC failed with error ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1293: CHIP Error 0x00000003: Incorrect state[0m
```
and looking at line 1293, which is inside `DeviceCommissioner::OnOperationalCertificateAddResponse` on that sha (`bc5129`) it's:
```
    VerifyOrExit(commissioner->mDeviceBeingCommissioned != nullptr, err = CHIP_ERROR_INCORRECT_STATE);
```
Now why is `mDeviceBeingCommissioned` null?  I don't know.

@cecille any idea what could be going on here?

cecille, 2022-02-09:
Sorry - missed this one in the email flood. Is this still a thing that's happening?

cecille, 2022-02-25:
@woody-apple - is this still happening?

cecille, 2022-04-19:
We did find a race condition in the darwin tests, which was fixed in this commit: https://github.com/project-chip/connectedhomeip/pull/16882/commits/9388830c98918d6410d9a633bf2083715b4eea46. This looks like what would happen. As we have a potential fix, and there has been no activity on this bug for some time, I am closing this. Please re-open or start a new issue if this re-occurs.

closed 2022-04-19 by cecille