project-chip/connectedhomeip #14327: Secure pairing failure

woody-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4938653851?check_suite_focus=true

``
2022-01-25 15:29:36.769 ERROR   15:29:26.253 - PAIR OUT  : [1643124566252] [9303:55574] CHIP: [TOO] Secure Pairing Failed
1137
```

 #### Proposed Solution
Fix

pan-apple, 2022-01-26:
In this CI run, there seems to be a mix of two separate pairing procedures.

1. Commissioning is in progress, and the controller receives a response to AddNOC command.
2. The CASE session establishment is in progress.

The controller receives the response for AddNOC command, but due to 2, it's not in the state where it's waiting for the response. That causes it to error out with the following error

```
2022-01-25 15:29:36.769 ERROR   15:29:26.253 - PAIR OUT  : [1643124566252] [9303:55574] CHIP: [CTL] Add NOC failed with error ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1290: CHIP Error 0x00000003: Incorrect state
1136
2022-01-25 15:29:36.769 ERROR   15:29:26.253 - PAIR OUT  : [1643124566252] [9303:55574] CHIP: [TOO] Secure Pairing Failed
```

bzbarsky-apple, 2022-01-27:
This looks like the same issue as https://github.com/project-chip/connectedhomeip/issues/14325 per the description above.

closed 2022-02-09 by woody-apple