project-chip/connectedhomeip #14328: Test failure in Darwin (KVS?)

woody-apple, 2022-01-26:
#### Problem
Test failed here: https://github.com/project-chip/connectedhomeip/runs/4938140252?check_suite_focus=true

```
2022-01-25 14:54:30.029 ERROR   14:54:29.893 - APP  ERR  :     #1 0x10efc942f in chip::DeviceLayer::PersistedStorage::KeyValueStoreManagerImpl::_Put(char const*, void const*, unsigned long) KeyValueStoreManagerImpl.mm:241
1162
2022-01-25 14:54:30.030 ERROR   14:54:29.895 - APP  ERR  :     #2 0x10f119543 in chip::DeviceLayer::PersistedStorage::KeyValueStoreManager::Put(char const*, void const*, unsigned long) KeyValueStoreManager.h:222
1163
2022-01-25 14:54:30.030 ERROR   14:54:29.895 - APP  ERR  :     #3 0x10f118c8c in chip::Server::ServerStorageDelegate::SyncSetKeyValue(char const*, void const*, unsigned short) Server.h:117
1164
2022-01-25 14:54:30.030 ERROR   14:54:29.895 - APP  ERR  :     #4 0x10f119182 in chip::Server::ServerStorageDelegate::SyncStore(unsigned char, char const*, void const*, unsigned short) Server.h:131
1165
2022-01-25 14:54:30.030 ERROR   14:54:29.895 - APP  ERR  :     #5 0x10f119378 in non-virtual thunk to chip::Server::ServerStorageDelegate::SyncStore(unsigned char, char const*, void const*, unsigned short) Server.h
1166
2022-01-25 14:54:30.030 ERROR   14:54:29.895 - APP  ERR  :     #6 0x10f03fb29 in chip::FabricInfo::CommitToStorage(chip::FabricStorage*) FabricTable.cpp:108
1167

```
 #### Proposed Solution
Fix

bzbarsky-apple, 2022-01-27:
Yes, this is the AddNOC crash under KVS that's also in:
* https://github.com/project-chip/connectedhomeip/issues/14329

turon, 2022-01-31:
Reliable commissioning on Darwin is critical for v1.
Removing `v1_triage_split_4` tag.

vivien-apple, 2022-02-09:
Is it still happening now that the KVS is properly accessing the store under the right dispatch queue ?

vivien-apple, 2022-03-30:
Closing. Please re-open if that still happens.

closed 2022-03-30 by vivien-apple