project-chip/connectedhomeip #14330: Secure pairing failed on Darwin

woody-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4922066894?check_suite_focus=true

```
2022-01-24 13:39:57.720 ERROR   13:39:57.447 - PAIR OUT  : [1643031597442] [9757:68401] CHIP: [CTL] Device failed to receive the operational certificate Response: 0x01
1120
2022-01-24 13:39:57.720 ERROR   13:39:57.447 - PAIR OUT  : [1643031597442] [9757:68401] CHIP: [TOO] Secure Pairing Failed
1121
2022-01-24 13:39:57.720 ERROR   13:39:57.447 - PAIR OUT  : [1643031597446] [9757:68401] CHIP: [TOO] Pairing Failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1418: CHIP Error 0x000000AC: Internal error
1122
2022-01-24 13:39:57.720 ERROR   13:39:57.447 - PAIR OUT  : [1643031597447] [9757:68401] CHIP: [DMG] ICR moving to [AwaitingDe]
1123
2022-01-24 13:39:57.720 ERROR   13:39:57.447 - PAIR OUT  : [1643031597447] [9757:68397] CHIP: [-] ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1418: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
1124
2022-01-24 13:39:57.721 ERROR   13:39:57.486 - PAIR OUT  : [1643031597447] [9757:68397] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1418: CHIP Error 0x000000AC: Internal error
1125
```
 #### Proposed Solution
Fix

pan-apple, 2022-01-26:
Looks like the device is crashing during processing of `AddNOC` command

```
2022-01-24 13:39:57.714 ERROR   13:39:45.453 - APP  ERR  : AddressSanitizer:DEADLYSIGNAL
1072
2022-01-24 13:39:57.714 ERROR   13:39:45.453 - APP  ERR  : =================================================================
1073
2022-01-24 13:39:57.714 ERROR   13:39:45.454 - APP  ERR  : ==9756==ERROR: AddressSanitizer: SEGV on unknown address 0x000125000050 (pc 0x7fff2044a41f bp 0x70000e3d32f0 sp 0x70000e3d3058 T2)
```

woody-apple, 2022-01-27:
Another:

woody-apple, 2022-01-27:
[logs_731128.zip](https://github.com/project-chip/connectedhomeip/files/7947728/logs_731128.zip)
Adding logs @pan-apple

woody-apple, 2022-01-27:
(Seen 4 tonight)

bzbarsky-apple, 2022-01-27:
Fwiw, this is the typical failrure I have been getting on https://github.com/project-chip/connectedhomeip/pull/14342 as well.  Typically the stack under AddNOC ends up in the KVS bits where ASAN then kills things because it doesn't like what's happening.

pan-apple, 2022-01-27:
> Another:
> https://github.com/project-chip/connectedhomeip/runs/4961578317?check_suite_focus=true

This one seems like a different failure
```
2022-01-27 04:57:43.917 ERROR   04:57:38.982 - APP  OUT  : [1643259458982] [62509:165733] CHIP: [IN] Sending encrypted msg 0x10e3f8810 with MessageCounter:14836182 to 0x000000000001B669 (1) at monotonic time: 1900277 msec
4422
2022-01-27 04:57:43.917 ERROR   04:57:41.009 - APP  OUT  : [1643259461007] [62509:164977] CHIP: [EM] Failed to Send CHIP MessageCounter:14836182 on exchange 27878r sendCount: 3 max retries: 3
```

woody-apple, 2022-02-09:
This was identified to be cross talk, and resolved.

closed 2022-02-09 by woody-apple

yqian442, 2022-07-11:
> This was identified to be cross talk, and resolved.

Hi, I also met the same problem. Can you explain how to solve it? Thank you.