project-chip/connectedhomeip #14338: Add persistent storage support for DefaultOTAProviders attribute

carol-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/issues/13855 will add support for reading/writing of DefaultOTAProviders attribute. However, because persistent storage is not supported for AttributeAccessInterface, this attribute will not be persisted.

 #### Proposed Solution
Maybe use the app-provided attr store with a key based on the path

"app provided attr store" == "app provided storage delegate that the attr store uses to store stuff"

yufengwangca, 2022-02-01:
You can set the storage type of the attribute to 'NVM', then the attribute should be persisted.

carol-apple, 2022-02-06:
@yufengwangca This particular attribute is a list of structs and so I had to set it to `External` and add support for it directly through AttributeAccessInterface. AFAIK, persistent storage for `External` is up to the implementation for AttributeAccessInterface. Unless there's something else I'm missing?

hrajendran-samsung, 2022-02-07:
Based on asking/looking around, it seems that the way to go would be to use the DefaultAttributePersistenceProvider. The attribute values ended up being stored in /tmp/chip_kvs , bas64 encoded I think.

hrajendran-samsung, 2022-02-08:
@Damian-Nordic suggested the following:

An implementation would be provided by the application but we could provide GenericOTARequestorStorage which would implement the methods using KeyValueStoreMgr().XXX functions

```
class OTARequestorStorage
{
virtual CHIP_ERROR StoreDefaultProviderList(...) = 0;
virtual CHIP_ERROR LoadDefaultProviderList(...) = 0;

virtual CHIP_ERROR StoreCurrentProvider(...) = 0;
virtual CHIP_ERROR LoadCurrentProvider(...) = 0;

virtual CHIP_ERROR StoreUpdateToken(...) = 0;
virtual CHIP_ERROR LoadUpdateToken(...) = 0;

...
};

```

PR #15970 (2022-03-09): [ota] Store Default OTA Providers in flash
#### Problem
OTA Requestor's `DefaultOTAProviders` attribute is not persisted.

#### Change overview
Add `OTARequestorStorage` to all examples using `OTARequestor`.
Use `OTARequestorStorage` interface to load `DefaultOTAProviders` on application startup and store it on each modification.
The current approach may cause some unnecessary writes since each attribute write is actually "Remove all" + "Append item" operations, but the existing attribute accessor interface doesn't notify an implementation that a given list operation is the last one in the act [... cut]
closed 2022-03-09 by Damian-Nordic