project-chip/connectedhomeip #14339: OTA Requestor needs to support querying providers from multiple fabrics

carol-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/issues/13782 will add support to periodic query using the `DefaultOTAProviders` attribute. The `DefaultOTAProviders` attribute is a list of default providers for each fabric that the requestor has been commissioned on. Currently, only one provider will ever be queried at a given time. Based on that assumption, there should be solution on which provider/fabric to query whenever the periodic timer fires.

 #### Proposed Solution
One solution could be that if a CASE session cannot be established with a particular provider, then the provider at the next fabric in the list should be queried. Perhaps a cache of which providers should be skipped/ignored needs to be maintained to support this.

There could be more complicated solutions such as using the `status` field of the QueryImageResponse to determine if a different provider should be used for query.

holbrookt, 2022-01-31:
Carol is this necessary for 1.0?

The reference Requestor API will still change its Provider target when it receives `AnnounceOTAProvider`, or when the `DefaultOTAProviders` changes, right?

carol-apple, 2022-02-16:
@holbrookt So what this issue is trying to capture is how to handle errors with querying, specifically, if the `DefaultOTAProviders` attribute contains multiple entries (one default OTA provider for each fabric). Current logic just finds the first entry and tries to query. But if for some reason, that provider is offline (assuming for longer than 24 hours), we should move on to the next entry. Right now, that will never happen.

holbrookt, 2022-02-17:
Seems necessary for 1.0 then - do you agree? There should be some backup for if the first try fails

carol-apple, 2022-02-17:
Yes correct, definitely needed for v1. This change can be worked on after https://github.com/project-chip/connectedhomeip/issues/13782 is complete as the logic will need to be factored in during the periodic query attempts.

closed 2022-03-09 by andy31415