project-chip/connectedhomeip #14359: Android failing CI

woody-apple, 2022-01-26:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4956765728?check_suite_focus=true

```
INFO    "Install SDK Patch Applier v4 (revision: 1)" complete.
3550
INFO    "Install SDK Patch Applier v4 (revision: 1)" finished.
3551
INFO    Checking the license for package NDK (Side by side) 21.4.7075529 in /opt/android/sdk/licenses
3552
INFO    License for package NDK (Side by side) 21.4.7075529 accepted.
3553
INFO    Preparing "Install NDK (Side by side) 21.4.7075529 (revision: 21.4.7075529)".
3554
INFO    Warning: An error occurred while preparing SDK package NDK (Side by side) 21.4.7075529: archive is not a ZIP archive.:
3555
INFO    java.util.zip.ZipException: archive is not a ZIP archive
3556
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.positionAtEndOfCentralDirectoryRecord(ZipFile.java:871)
3557
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.positionAtCentralDirectory(ZipFile.java:801)
3558
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.populateFromCentralDirectory(ZipFile.java:542)
3559
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.<init>(ZipFile.java:217)
3560
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.<init>(ZipFile.java:193)
3561
INFO    	at org.apache.commons.compress.archivers.zip.ZipFile.<init>(ZipFile.java:154)
3562
INFO    	at com.android.repository.util.InstallerUtil.unzip(InstallerUtil.java:98)
3563
```
 #### Proposed Solution
Fix

woody-apple, 2022-01-27:
Another:
https://github.com/project-chip/connectedhomeip/runs/4960482747?check_suite_focus=true

woody-apple, 2022-01-27:
Another: https://github.com/project-chip/connectedhomeip/runs/4969793252?check_suite_focus=true

austinh0, 2022-01-27:
That last run is an actual build error.

```
WARNING e: /__w/connectedhomeip/connectedhomeip/src/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/MultiAdminClientFragment.kt: (102, 53): None of the following functions can be called with the arguments supplied: 
WARNING public open fun revokeCommissioning(p0: ChipClusters.DefaultClusterCallback!, p1: Int): Unit defined in chip.devicecontroller.ChipClusters.AdministratorCommissioningCluster
WARNING private open fun revokeCommissioning(p0: Long, p1: ChipClusters.DefaultClusterCallback!, p2: Int?): Unit defined in chip.devicecontroller.ChipClusters.AdministratorCommissioningCluster
```

andy31415, 2022-02-07:
I believe this may be infra flakyness. Did not see it in a while.

woody-apple, 2022-02-08:
Then let's close it

closed 2022-02-08 by woody-apple