project-chip/connectedhomeip #14366: Investigation: in particular, the way things are right now any events generated by a PASE session before it's picked up a fabric association will be visible to all consumers.

yunhanw-google, 2022-01-26:
So in particular, the way things are right now any events generated by a PASE session before it's picked up a fabric association will be visible to all consumers.

Maybe that's not a problem in practice because you can't add ACLS via a PASE session before it picks up a fabric?  But we would need to pretty carefully check the ordering of things in AddNOC....

On the other hand, if we allowed event.mFabricIndex to have the "no fabric" value (0).... I guess a later PASE session would be able to see the events, which is not desirable?

Maybe the right answer is that fabric-scoped events emitted not in the context of a fabric should just be dropped completely?  Worth checking with @mlepage-google and @tcarmelveilleux whether they have thoughts on this.  But we can also adjust it as a followup.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/13907#discussion_r793105854_

bzbarsky-apple, 2022-01-26:
@mrjerryjohns @anush-apple

bzbarsky-apple, 2022-01-27:
Needs to be v1.0, because this is a security problem if we get this wrong.

bzbarsky-apple, 2022-06-13:
Removing "deferral candidate".  This exposes data across fabrics in ways that are not OK, last I checked.

closed 2022-08-17 by yunhanw-google