project-chip/connectedhomeip #14369: [TC_MC_3.6] Missing `CurrentApp` attribute in `applicationlauncher` cluster in TE7.5

liamgonyea, 2022-01-27:
#### Problem
CurrentApp attribute is not listed in the available attributes but is needed to run `[TC-MC-3.6] Current App Verification`, step 1.

[image]

#### Test Case
TE7.5 commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`
3.4.2. [TC-MC-3.6] Current App Verification
https://github.com/CHIP-Specifications/chip-test-plans/blob/TE_7_5_updates_IDM/src/cluster/media.adoc#tc-mc-3-6-current-app-verification

#### Logs

> 16-MacBook-Pro:chip-tool liam$ ./out/debug/chip-tool applicationlauncher read 
> [1642811926723] [99229:4621821] CHIP: [TOO] Missing attribute name
> Usage:
>   ./out/debug/chip-tool applicationlauncher read attribute-name [param1 param2 ...]
> 
>   +-------------------------------------------------------------------------------------+
>   | Attributes:                                                                         |
>   +-------------------------------------------------------------------------------------+
>   | * application-launcher-list                                                         |
>   | * attribute-list                                                                    |
>   | * cluster-revision                                                                  |
>   +-------------------------------------------------------------------------------------+
> [1642811926723] [99229:4621821] CHIP: [TOO] Run command failure: ../../commands/common/Commands.cpp:99: Error 0x0000002F

yufengwangca, 2022-02-07:
@lazarkov I saw you are working on the initial implementation of this cluster, could we enable this attribute and complete the remaining work if there is any

lazarkov, 2022-02-07:
@yufengwangca yes there are 2 PRs that will resolve this:
https://github.com/project-chip/connectedhomeip/pull/14794
https://github.com/project-chip/connectedhomeip/pull/14841

lazarkov, 2022-02-07:
Either one of them will, and probably both of them will be merged in the next 24h.

lazarkov, 2022-02-08:
Resolved.

closed 2022-02-08 by lazarkov