project-chip/connectedhomeip #14370: ExchangeMgr.cpp:300: CHIP Error 0x0000002A: Invalid message

woody-apple, 2022-01-27:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4960771205?check_suite_focus=true

```
2022-01-27 02:58:36.083 ERROR   02:58:04.999 - APP  OUT  : [1643252284998] [9956:60530] CHIP: [EM] OnMessageReceived failed, err = ../../examples/all-clusters-app/linux/third_party/connectedhomeip/src/messaging/ExchangeMgr.cpp:300: CHIP Error 0x0000002A: Invalid message type
967
2022-01-27 02:58:36.083 ERROR   02:58:06.998 - PAIR OUT  : [1643252286997] [9957:60541] CHIP: [EM] Failed to Send CHIP MessageCounter:2422476098 on exchange 44019i sendCount: 3 max retries: 3
968
2022-01-27 02:58:36.083 ERROR   02:58:36.001 - PAIR OUT  : [1643252316000] [9957:60649] CHIP: [TOO] Pairing Failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1397: CHIP Error 0x00000032: Timeout
969
2022-01-27 02:58:36.083 ERROR   02:58:36.001 - PAIR OUT  : [1643252316001] [9957:60536] CHIP: [-] ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1397: CHIP Error 0x00000032: Timeout at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
970
2022-01-27 02:58:36.083 ERROR   02:58:36.001 - PAIR OUT  : [1643252316001] [9957:60536] CHIP: [TOO] Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1397: CHIP Error 0x00000032: Timeout
971
```
 #### Proposed Solution
<suggested fix, suggested enhancement>

woody-apple, 2022-01-27:
Another: https://github.com/project-chip/connectedhomeip/runs/4960771205?check_suite_focus=true

Morozov-5F, 2022-01-27:
Looks really similar: https://github.com/project-chip/connectedhomeip/runs/4971261703?check_suite_focus=true

woody-apple, 2022-01-27:
https://github.com/project-chip/connectedhomeip/runs/4971261703?check_suite_focus=true

woody-apple, 2022-02-09:
This was resolved

closed 2022-02-09 by woody-apple