project-chip/connectedhomeip #14374: Non-success NOCResponse fails parse / dispatch

msandstedt, 2022-01-27:
#### Problem

NOCResponse with status code 3 (non-success) and no fabric index enclosed is not successfully passed to the on-failure callback:

```
[1643258092.739042][2486631:2486636] CHIP:DMG: InvokeResponseMessage =
[1643258092.739068][2486631:2486636] CHIP:DMG: {
[1643258092.739118][2486631:2486636] CHIP:DMG: 	suppressResponse = false, 
[1643258092.739166][2486631:2486636] CHIP:DMG: 	InvokeResponseIBs =
[1643258092.739205][2486631:2486636] CHIP:DMG: 	[
[1643258092.739230][2486631:2486636] CHIP:DMG: 		InvokeResponseIB =
[1643258092.739286][2486631:2486636] CHIP:DMG: 		{
[1643258092.739320][2486631:2486636] CHIP:DMG: 			CommandDataIB =
[1643258092.739355][2486631:2486636] CHIP:DMG: 			{
[1643258092.739384][2486631:2486636] CHIP:DMG: 				CommandPathIB =
[1643258092.739432][2486631:2486636] CHIP:DMG: 				{
[1643258092.739467][2486631:2486636] CHIP:DMG: 					EndpointId = 0x0,
[1643258092.739500][2486631:2486636] CHIP:DMG: 					ClusterId = 0x3e,
[1643258092.739537][2486631:2486636] CHIP:DMG: 					CommandId = 0x8,
[1643258092.739568][2486631:2486636] CHIP:DMG: 				},
[1643258092.739605][2486631:2486636] CHIP:DMG: 				
[1643258092.739629][2486631:2486636] CHIP:DMG: 				CommandData = 
[1643258092.739666][2486631:2486636] CHIP:DMG: 				{
[1643258092.739704][2486631:2486636] CHIP:DMG: 					0x0 = 3, 
[1643258092.739744][2486631:2486636] CHIP:DMG: 					0x2 = "", 
[1643258092.739780][2486631:2486636] CHIP:DMG: 				},
[1643258092.739805][2486631:2486636] CHIP:DMG: 			},
[1643258092.739843][2486631:2486636] CHIP:DMG: 			
[1643258092.739863][2486631:2486636] CHIP:DMG: 		},
[1643258092.739902][2486631:2486636] CHIP:DMG: 		
[1643258092.739926][2486631:2486636] CHIP:DMG: 	],
[1643258092.739963][2486631:2486636] CHIP:DMG: 	
[1643258092.739990][2486631:2486636] CHIP:DMG: },
[1643258092.740075][2486631:2486636] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0008
[1643258092.740110][2486631:2486636] CHIP:ZCL: Failed to dispatch command, 2/3 arguments parsed, TLVError=../zzz_generated/controller-clusters/zap-generated/IMClusterCommandHandler.cpp:2416: Success, UnpackError=../zzz_generated/controller-clusters/zap-generated/IMClusterCommandHandler.cpp:2416: Success (last decoded tag = 2
```

From the spec, it is correct that fabric index is not enclosed for non-success:

> FabricIndex Field
> This field SHALL be present whenever StatusCode has a value of Ok. If present, it SHALL contain the Fabric Index of the Fabric last added, removed or updated.

 #### Proposed Solution

Make the parser accept FabricIndex missing for the non-success case.  If this is unworkable, we could instead enclose the 0-no-fabric-index for the non-success case, but this also requires a spec change.

bzbarsky-apple, 2022-06-13:
@msandstedt Is this still an issue?

msandstedt, 2022-06-13:
No, this works correctly now:

```
[1655152985.931712][3382211:3382216] CHIP:DMG: InvokeResponseMessage =
[1655152985.931719][3382211:3382216] CHIP:DMG: {
[1655152985.931726][3382211:3382216] CHIP:DMG: 	suppressResponse = false, 
[1655152985.931731][3382211:3382216] CHIP:DMG: 	InvokeResponseIBs =
[1655152985.931746][3382211:3382216] CHIP:DMG: 	[
[1655152985.931754][3382211:3382216] CHIP:DMG: 		InvokeResponseIB =
[1655152985.931772][3382211:3382216] CHIP:DMG: 		{
[1655152985.931781][3382211:3382216] CHIP:DMG: 			CommandDataIB =
[1655152985.931794][3382211:3382216] CHIP:DMG: 			{
[1655152985.931804][3382211:3382216] CHIP:DMG: 				CommandPathIB =
[1655152985.931812][3382211:3382216] CHIP:DMG: 				{
[1655152985.931824][3382211:3382216] CHIP:DMG: 					EndpointId = 0x0,
[1655152985.931834][3382211:3382216] CHIP:DMG: 					ClusterId = 0x30,
[1655152985.931844][3382211:3382216] CHIP:DMG: 					CommandId = 0x1,
[1655152985.931854][3382211:3382216] CHIP:DMG: 				},
[1655152985.931868][3382211:3382216] CHIP:DMG: 				
[1655152985.931877][3382211:3382216] CHIP:DMG: 				CommandFields = 
[1655152985.931884][3382211:3382216] CHIP:DMG: 				{
[1655152985.931901][3382211:3382216] CHIP:DMG: 					0x0 = 4, 
[1655152985.931913][3382211:3382216] CHIP:DMG: 					0x1 = "", 
[1655152985.931923][3382211:3382216] CHIP:DMG: 				},
[1655152985.931932][3382211:3382216] CHIP:DMG: 			},
[1655152985.931945][3382211:3382216] CHIP:DMG: 			
[1655152985.931953][3382211:3382216] CHIP:DMG: 		},
[1655152985.931966][3382211:3382216] CHIP:DMG: 		
[1655152985.931973][3382211:3382216] CHIP:DMG: 	],
[1655152985.931994][3382211:3382216] CHIP:DMG: 	
[1655152985.932004][3382211:3382216] CHIP:DMG: 	InteractionModelRevision = 1
[1655152985.932014][3382211:3382216] CHIP:DMG: },
[1655152985.932066][3382211:3382216] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
[1655152985.932091][3382211:3382216] CHIP:CTL: Failsafe disarmed
[1655152985.932110][3382211:3382216] CHIP:CTL: Successfully finished commissioning step 'Cleanup'
[1655152985.932134][3382211:3382216] CHIP:TOO: Device commissioning Failure: ../src/controller/CHIPDeviceController.cpp:1179: CHIP Error 0x00000078: CaseAdminSubject is not valid
[1655152985.932164][3382211:3382216] CHIP:DIS: Closing all BLE connections
[1655152985.932177][3382211:3382216] CHIP:IN: Clearing BLE pending packets.
[1655152985.932215][3382211:3382216] CHIP:BLE: Auto-closing end point's BLE connection.
[1655152985.932224][3382211:3382216] CHIP:DL: Closing BLE GATT connection (con 0x7f8164030f80)
[1655152985.932280][3382211:3382216] CHIP:DMG: ICR moving to [AwaitingDe]
[1655152985.932418][3382211:3382214] CHIP:DL: BluezDisconnect peer=98:F4:AB:6B:36:7E
[1655152985.932465][3382211:3382211] CHIP:CTL: Shutting down the commissioner
[1655152985.932487][3382211:3382211] CHIP:CTL: Shutting down the controller
[1655152985.932501][3382211:3382211] CHIP:CTL: Shutting down the commissioner
[1655152985.932505][3382211:3382211] CHIP:CTL: Shutting down the controller
[1655152985.932511][3382211:3382211] CHIP:IN: Expiring all connections for fabric 1!!
[1655152985.932536][3382211:3382211] CHIP:CTL: Shutting down the commissioner
[1655152985.932546][3382211:3382211] CHIP:CTL: Shutting down the controller
[1655152985.932551][3382211:3382211] CHIP:IN: Expiring all connections for fabric 2!!
[1655152985.932563][3382211:3382211] CHIP:CTL: Shutting down the commissioner
[1655152985.932570][3382211:3382211] CHIP:CTL: Shutting down the controller
[1655152985.932575][3382211:3382211] CHIP:IN: Expiring all connections for fabric 3!!
[1655152985.932589][3382211:3382211] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1655152985.932708][3382211:3382211] CHIP:DMG: IM WH moving to [Uninitialized]
[1655152985.932716][3382211:3382211] CHIP:DMG: IM WH moving to [Uninitialized]
[1655152985.932720][3382211:3382211] CHIP:DMG: IM WH moving to [Uninitialized]
[1655152985.932724][3382211:3382211] CHIP:DMG: IM WH moving to [Uninitialized]
[1655152985.932730][3382211:3382211] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1655152985.932751][3382211:3382211] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1655152985.932904][3382211:3382211] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-3oVat9)
[1655152985.933157][3382211:3382211] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1655152985.933183][3382211:3382211] CHIP:DL: NVS set: chip-counters/total-operational-hours = 4 (0x4)
[1655152985.933193][3382211:3382211] CHIP:DL: Inet Layer shutdown
[1655152985.933201][3382211:3382211] CHIP:DL: BLE shutdown
[1655152985.933590][3382211:3382211] CHIP:DL: System Layer shutdown
[1655152985.933644][3382211:3382211] CHIP:IN: SecureSession[0x5634760b0510]: Released - Type:1 LSID:26266
[1655152985.933686][3382211:3382211] CHIP:TOO: Run command failure: ../src/controller/CHIPDeviceController.cpp:1179: CHIP Error 0x00000078: CaseAdminSubject is not valid
```

closed 2022-06-13 by msandstedt