project-chip/connectedhomeip #14379: ESP32 Build failed

woody-apple, 2022-01-27:
#### Problem
https://github.com/project-chip/connectedhomeip/runs/4961781714?check_suite_focus=true

```
[1/1205] Generating project_elf_src_esp32.c
598
[2/1205] Generating ../../ota_data_initial.bin
599
[3/1205] To flash /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/chip-lighting-app.bin run ./build/chip-lighting-app.flash.py
600
FAILED: CMakeFiles/flashing_script 
601
cd /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build && /__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/bin/python /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/../../../scripts/flashing/gen_flashing_script.py esp32 --output /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/chip-lighting-app.flash.py --port "" --baud 460800 --before default_reset --after hard_reset --application chip-lighting-app.bin --bootloader bootloader/bootloader.bin --partition partition_table/partition-table.bin --use-partition-file /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/partition_table/partition-table.bin --use-parttool /opt/espressif/esp-idf/components/partition_table/parttool.py --use-sdkconfig /__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/sdkconfig
602
Traceback (most recent call last):
603
  File "/__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/../../../scripts/flashing/gen_flashing_script.py", line 24, in <module>
604
    sys.exit(platform.Flasher().make_wrapper(sys.argv))
605
  File "/__w/connectedhomeip/connectedhomeip/scripts/flashing/esp32_firmware_utils.py", line 389, in make_wrapper
606
    super().make_wrapper(argv)
607
  File "/__w/connectedhomeip/connectedhomeip/scripts/flashing/firmware_utils.py", line 404, in make_wrapper
608
    self._platform_wrapper_args(args)
609
  File "/__w/connectedhomeip/connectedhomeip/scripts/flashing/esp32_firmware_utils.py", line 402, in _platform_wrapper_args
610
    args.application_offset = self.get_partition_info(
611
  File "/__w/connectedhomeip/connectedhomeip/scripts/flashing/esp32_firmware_utils.py", line 370, in get_partition_info
612
    return self.run_tool(
613
  File "/__w/connectedhomeip/connectedhomeip/scripts/flashing/firmware_utils.py", line 217, in run_tool
614
    result = subprocess.run(
615
  File "/usr/lib/python3.9/subprocess.py", line 528, in run
616
    raise CalledProcessError(retcode, process.args,
617
subprocess.CalledProcessError: Command '['/opt/espressif/esp-idf/components/partition_table/parttool.py', '--partition-table-file', '/__w/connectedhomeip/connectedhomeip/examples/lighting-app/esp32/build/partition_table/partition-table.bin', 'get_partition_info', '--partition-boot-default', '--info', 'offset']' returned non-zero exit status 1.
618
```

 #### Proposed Solution
Fix

dhrishi, 2022-01-27:
@woody-apple Looks like a one-off failure? On `c62489bc08709e2932cc20565c7a133665d16291` I am able to compile `lighting-app/esp32` locally. Also, I could successfully run `scripts/examples/esp_example.sh lighting-app sdkconfig.defaults` locally

woody-apple, 2022-01-27:
@dhrishi here's another: https://github.com/project-chip/connectedhomeip/runs/4968514674?check_suite_focus=true

dhrishi, 2022-01-29:
@woody-apple Ohh okay. I will check

turon, 2022-01-31:
All platforms must build for v1, so removing `v1_triage_split_4` tag.

woody-apple, 2022-02-09:
Haven't seen recently, moving to v1.X

closed 2022-10-09 by stale[bot]