project-chip/connectedhomeip #14395: [TC-DA-1.2]chip-tool-Device Attestation Request Validation not supported

sumaky, 2022-01-27:
Problem:
1.The logs obtained from CHIP tool does not allow to validate the following:

- AttestationElements
- certification_declaration elements 

2. The data in the logs are not fully visible for CD

Test Plan Link:
(https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/deviceattestation.adocl)

Proposed Solution:

- The logs should have entire data for the verification of the conditions mentioned in the test plan
- Provision of programmatic interfaces for the Validation of AttestationElements / CD

[CONT_TC-DA.txt](https://github.com/project-chip/connectedhomeip/files/7949846/CONT_TC-DA.txt)

tcarmelveilleux, 2022-03-25:
Logs are not the way to determine CD and AttestationElements validity. The elements can only be validated with some automation, since there are too many rules to reliably determine the outcome purely by inspection, and making the software log in a human readable way all the traffic is unwieldy.

The approach I suggested in the past to @vivien-apple to support this is to have the YAML framework be able to save/store a given command response and some attribute values previously read (e.g. the device attestation certificate chain), and then run a named tool or procedure on the saved data. This would allow using some of the tools we already have (such as chip-cert) to be expanded to do the necessary validation and provide a report, without changing SDK core code.

The approach is also compatible with `matter-repl` which is already able to store payloads if necessary, but is not yet used by the test harness.

The above would work for test cases that relate both to AttestationRequest/Response, and CSRRequest/Response and AddNOC test cases.

vivien-apple, 2022-04-13:
My understanding of this issue is that we need 2 things:
  * #16846
  * Have a script to decode the payload of the messages that we are interested in

Is there anything else that I missed ?

sumaky, 2022-04-14:
@vivien-apple I am not sure if this question is addressed to me. If so please attach a example log for device attestation and script so that we are able to analyze the output and do the logical mapping with the test plan expected result

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning with @vivien-apple

franck-apple, 2022-06-08:
should be fixed by Vivien's PR #18893. @sumaky can you please verify and close this issue?
Otherwise, please let @vivien-apple know if something is missing.