project-chip/connectedhomeip #14397: [TC-DA-1.3]chip-tool-Device Attestation Request Validation [DUT Commissioner] is not supported

sumaky, 2022-01-27:
#### Problem

1. To validate the device attestation procedure with DUT as commissioner is not supported in CHIP tool
2. Attestation NONCE validation is not supported

 #### Proposed Solution

1. Provision of API's to support  device attestation by enabling DUT as commissioner 
2. Method to verify DUT generates unique nonce each time

Test Plan Link
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/deviceattestation.adoc

vivien-apple, 2022-04-13:
https://github.com/project-chip/connectedhomeip/pull/17107 has landed to provide a way for placeholder apps (or the all-clusters-app) to customize the responses in order to trigger errors.
What we need now is correct values to use and pass to the testing apps.

Let me know if there is anything else to do here.

vivien-apple, 2022-04-13:
My understanding of this issue is that we need 2 things:
  * #16846
  * Have a script to decode the payload of the messages that we are interested in

Is there anything else that I missed ?

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @vivien-apple

franck-apple, 2022-06-08:
should be fixed by Vivien's PR #18893. @sumaky can you please verify and close this issue?
Otherwise, please let @vivien-apple know if something is missing.

vivien-apple, 2022-06-09:
I would assume that this is done. It I'm wrong please reopen a separate issue with the specific parts that are missing. thanks!