project-chip/connectedhomeip #14399: [TC-DA-1.4]chip-tool-TC-DA-1.4 Device Attestation Request Validation-Error Scenarios[DUT-Commissioner] support required

sumaky, 2022-01-27:
#### Problem
In the current chip tool inducing error conditions is not possible.

DUT as Commissioner has to be tested with multiple error scenarios as follows:

1. Commissionee is not yet certified
2. Commissionee’s PAA is not in the Commissioner’s trusted set.
3. Commissionee provides an invalid DAC and PAI chain as validated by TH against the RFC5280 certificate Basic path validation algorithm
4. Commissionee provides DAC with validity period that is not in range of PAA/PAI
5. Certification Declaration has only one of the following fields dac_origin_vendor_id dac_origin_product_id
6. vendor_id field in the Certification Declaration does not match with the VendorID attribute found in the Basic Information cluster
7. The Vendor ID in the DAC does not match the vendor_id field in the Certification Declaration.
8. The Product ID (PID) the DAC is not present in the product_id_array field in the Certification Declaration

 #### Proposed Solution

Provision of API's for creating the above error scenarios for testing the DUT

Test Plan Link:
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/deviceattestation.adoc

tcarmelveilleux, 2022-03-25:
As mentioned in #14397 , this is not a chip-tool issue: Commissioner as DUT does not involve chip-tool since chip-tool has the role of Commissioner and not device.

I would argue this issue is 100% a duplicate of #14397

vivien-apple, 2022-04-13:
https://github.com/project-chip/connectedhomeip/pull/17107 has landed to provide a way for placeholder apps (or the all-clusters-app) to customize the responses in order to trigger errors.
What we need now is correct values to use and pass to the testing apps.

vivien-apple, 2022-04-13:
Please re-open if the provided API is not enough.

closed 2022-04-13 by vivien-apple