project-chip/connectedhomeip #14403: Need factory reset command for SDK

Survensa, 2022-01-27:
Summary :  Once we send the factory reset command, the device should delete all previous provisioning information. 

Based on LocalConfigDisabled Attribute

The LocalConfigDisabled attribute SHALL allow a local Node configuration to be disabled. When
this attribute is set to True the Node SHALL disable the ability to configure the Node through an onNode user interface. The value of the LocalConfigDisabled attribute SHALL NOT in any way modify,
disable, or otherwise affect the user’s ability to trigger a factory reset on the Node.

Refer : 11.1.6.1. Attributes https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/build-sample/pdf/main.pdf

bzbarsky-apple, 2022-01-27:
> Summary : Once we send the factory reset command, the device should delete all previous provisioning information.

What factory reset command?  Matter has no such command, @Survensa.

> The value of the LocalConfigDisabled attribute SHALL NOT in any way modify, disable, or otherwise affect the user’s ability to trigger a factory reset on the Node.

What this means is this.  If the device has a "press and hold this to factory reset" button, that button will continue to function as it should even if LocalConfigDisabled is set to false.

yufengwangca, 2022-02-01:
@Survensa 

Are you talking about device specific factory reset command? We don't have such command specified in the cluster.

cjandhyala, 2022-02-02:
@bzbarsky-apple @yufengwangca What @Survensa meant here, 
1. Make sure SDK has the implementation that spec recommended for factory reset condition , Ref attached
2. Need a way to send that command to the example apps, Either by means of some signal or something,  so that we can test device is performing all the actions that spec is recommending when factory reset command sent

[image]

bzbarsky-apple, 2022-02-02:
@cjandhyala Just to be clear: that section of the spec is not about a "command" in the Matter sense, right?

The SDK itself does not provide a factory reset mechanism, which involves wiping various storage that the application, not the SDK, knows about.  That mechanism needs to be provided by the SDK consumer.

Some of the in-tree example apps that run on actual devices do have such mechanisms.  That said, the mention of "some signal or something" sounds like you're talking about the linux/darwin apps, and for those "factory reset" corrresponds to killing the app, erasing the KVS on disk, and restarting the app...

cjandhyala, 2022-02-03:
@bzbarsky-apple Correct, that section of the spec is not about the command. 

I agree that the actual implementation of factory reset function should come from the consumer of the SDK.Just like the OTA feature has reference implementation, we were expecting there may be some reference implementation/API  that each platform owner adopts. So we wanted to make sure SDK has the spec required things implemented for factory reset condition.

if "factory reset" corrresponds to killing the app, erasing the KVS on disk, and restarting the app, we are good with it.

woody-apple, 2022-02-09:
Closing this, as there's no requirement here from the SDK.

closed 2022-02-09 by woody-apple