project-chip/connectedhomeip #14404: [TC-DA-1.5, 1.6]chip-tool- NOCSR Procedure Validation-Support Required

sumaky, 2022-01-27:
#### Problem
**Support Required to verify the following (DUT as Commissionee):**

1. DUT generates the NOCSR Information using CSRResponse Command
2. DUT generated NOCSR Information includes a signature using the Device Attestation Private Key
3. CSR SHALL follow the encoding and rules from PKCS #10
4. DUT generated that Node Operational Key Pair is unique
5. DUT rejects invalid CSRNonce sent by TH1
6. csr field is DER-encoded octet string
7. csr follows the encoding and rules from PKCS #10
8. CSRNonce field is OCTET string of length 32
9. CSRNonce value should match the value CSR Nonce field in the corresponding CSRRequest Command
10. Verify that NOCSRElement size should not be greater than RESP_MAX (900 bytes)

**Support required to inject the following error in NOCSR Information-CSRRequest Command to test DUT as Commissioner**

1. CSRNonce is OCTET string of length> 32
2. CSRNonce value does not match the value CSR Nonce field in the corresponding CSRRequest Command
3. NOCSRElement size is greater than RESP_MAX
4. Invalid NOCSR with an existing key pair 

Test Plan Link
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/deviceattestation.adoc

franck-apple, 2022-06-08:
First part (first checkbox) should be fixed by Vivien's PR #18893.
Second part to come.

vivien-apple, 2022-06-10:
> Most of second part is in #19461
> 
> It is just missing "Invalid NOCSR with an existing key pair" since I'm not sure of what it implies. @emargolis can you help me understand this part ?

I checked with @tcarmelveilleux what needs to be done. I will do it in a separate PR as it is a bit more involved than the previous cases.

krypton36, 2022-06-15:
Spoke to @vivien-apple offline and the steps to test "Invalid NOCSR with an existing key pair" CSRResponse is as follows:
```
1. Start all-clusters-app with cert_error_csr_existing_keypair
2. Pair all-clusters-app using chip-tool (KeyPair Generated)
3. Open a commissioning window using your preferred method
4. Pair all-clusters-app using chip-tool with a "different fabric" (Should Fail using existing KeyPair)
```

closed 2022-06-15 by vivien-apple