project-chip/connectedhomeip #14419: Add custom required privilege definitions

mlepage-google, 2022-01-27:
xxx

mlepage-google, 2022-04-01:
Reopening just so I have an issue to track adding the actual XML declarations for any remaining custom privileges. Normally this would be done by cluster authors, but it's easier overall for me to do them in one big swoop.

mlepage-google, 2022-04-06:
I think after #17115 and #17134 PRs it's really only the **pump control and configuration cluster** that needs updating; that one's waiting for a small spec change regarding the privileges (which currently require manage for read as well as write, but probably will be updated to only require manage for read).

mlepage-google, 2022-04-07:
Probably should also wait for this other pump configuration-and-control-cluster.xml change to land.
PR #17181

tlykkeberg-grundfos, 2022-04-07:
Will be fixed for pump-configuration-and-control-cluster in #17181

mlepage-google, 2022-04-07:
I do see @tlykkeberg-grundfos is handling the last access definitions for pump in #17181 so I will close this issue now. Thanks!

PR #16327 (2022-03-23): Add zap gen for Matter access privilege definitions
#### Problem

Data is not generated for access privileges.

Issue #14419

#### Change overview

New zap template iterates over access definitions for app server
clusters, for attributes/commands/events, to generate parallel arrays of
custom privileges for read/write attribute, invoke command, and read
event.

New privilege storage source files provide an API to access the generated
data, and an implementation using the generated data.

The data is generated and the storage is built per-app.

The library (DM, IM, app common) RequiredPrivilege  [... cut]
commit c7b49135 (2022-03-23): Add zap gen for Matter access privilege definitions (#16327)

New zap template iterates over access definitions for app server
clusters, for attributes/commands/events, to generate parallel arrays of
custom privileges for read/write attribute, invoke command, and read
event.

New privilege storage source files provide an API to access the generated
data, and an implementation using the generated data.

The data is generated and the storage is built per-app.

The library (DM, IM, app common) RequiredPrivilege module now uses the
privilege-storage API to access populated data on a per-app basis. [... cut]
closed 2022-04-07 by mlepage-google