project-chip/connectedhomeip #14424: change to return null once 13606 is fixed

woody-apple, 2022-01-27:
false

---

https://github.com/project-chip/connectedhomeip/blob/0c94a8093a1ff601fc3301aef586fabb5ffc2d70/src/app/clusters/application-launcher-server/application-launcher-server.cpp#L126-L131

---

###### This issue was generated by [todo-issue](https://github.com/DerJuulsn/todo-issue) based on a `TODO` comment in 0c94a8093a1ff601fc3301aef586fabb5ffc2d70. It&#x27;s been assigned to @chrisdecenzo because they committed the code.

turon, 2022-01-31:
Linking to referenced #13606

turon, 2022-01-31:
Passing EP as uint16_t, and properly applying nullable fields within a cluster looks like something that needs to be fixed for v1.
Removing tag `v1_triage_split_4`.

closed 2022-02-15 by lazarkov