project-chip/connectedhomeip #14438: Access control events should be fabric scoped

mlepage-google, 2022-01-27:
Events for access control cluster should be fabric scoped.

This will require a spec change, and because this may be the first instance of fabric scope/filtered events, a syntax may have to be defined. (I.e. no good example to copy.)

This also requires IM support for fabric filtered events. PR #13907 does that, so just need to validate and see if anything else needs to be done.

mlepage-google, 2022-01-31:
Spec PR
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/4889

turon, 2022-01-31:
Removing triage tag.  This bug is v1 worthy.

mlepage-google, 2022-01-31:
> Removing triage tag `v1_triage_split_4`. This bug is v1 worthy.

Yes this has to be done, it's no sense having fabric scoped/sensitive attributes but not fabric scoped/sensitive events which include the same content.

mlepage-google, 2022-01-31:
I tried this with Access Control.

In one REPL, I set up two controllers on fabric 1, one subscribed to this event on fabric 1.
In another REPL, I set up a controller on fabric 2, subscribed to this event on fabric 2.

In the first REPL, I wrote the ACL attribute with a new list value.
In the first REPL, I received event notifications for changed entries.
In the second REPL, I received no such notifications.

This appears to be working as intended.

closed 2022-01-31 by mlepage-google