project-chip/connectedhomeip #14444: TestPeerConnections fails if SecureSessionTable uses heap allocation

kpschoedel, 2022-01-27:
#### Problem

If `SecureSessionTable::mEntires` is a heap-allocated pool, `TestPeerConnections` fails:

```
'#0:','Transport-SecureSessionTable'
../../src/transport/tests/TestPeerConnections.cpp:94: assertion failed: "!optionalSession.HasValue()"
'#3:','BasicFunctionality','FAILED'
'#3:','FindByKeyId       ','PASSED'
../../src/transport/tests/TestPeerConnections.cpp:159: assertion failed: "!optionalSession.HasValue()"
../../src/transport/tests/TestPeerConnections.cpp:200: assertion failed: "callInfo.callCount == 1"
'#3:','ExpireConnections ','FAILED'
```

This is surprising and may point to underlying unsafeness.

 #### Proposed Solution

Investigate and resolve.

bzbarsky-apple, 2022-09-10:
@kghost @mrjerryjohns is this still an issue?

woody-apple, 2022-11-02:
Issue Scrub: Assigning to @kpschoedel to confirm if this is still happening.

kpschoedel, 2022-11-02:
This case can't be happening any more; the code is now using `ObjectPool`, so ASAN would catch any use-after-free.

closed 2022-11-02 by kpschoedel