project-chip/connectedhomeip #14446: TestReadInteraction aborts if ReliableMessageMgr::mRetransTable uses heap allocation

kpschoedel, 2022-01-27:
#### Problem

If `ReliableMessageMgr::mRetransTable` is a heap-allocated pool, `TestReadInteraction` aborts:

```
CHIP:SPT: VerifyOrDie failure at ../../src/lib/core/Optional.h:170: HasValue()
```

This is surprising and may indicate an underlying memory safety problem.

 #### Proposed Solution

Investigate and resolve.

kpschoedel, 2022-02-03:
`TestReadInteraction::TestReadChunking` leads to a use-after-free of a `RetransTableEntry`, which has been masked by the use of a static `BitMapObjectPool` for `ReliableMessageMgr::mRetransTable`.

Complicated details to follow.

kpschoedel, 2022-02-03:
Enter `TestReadInteraction::TestReadChunking(…)`
At 817: `InteractionModelEngine::GetInstance()->GetReportingEngine().Run();`

Enter `Engine::Run()`
At 503: `CHIP_ERROR err = BuildAndSendSingleReportData(readHandler);`

Enter `Engine::BuildAndSendSingleReportData(…)`

_CHIP:DMG: <RE:Run> Cluster fff10002, Attribute fff10004 is dirty_
_CHIP:DMG: Reading Mock Cluster fff10002, Field fff10004 is dirty_
_CHIP:DMG: Error retrieving data from clusterId: 0xFFF1_0002, err = ../../src/system/TLVPacketBufferBackingStore.cpp:88: Error 0x0000000B_
_CHIP:DMG: <RE:Run> We cannot put more chunks into this report. Enable chunking._
_CHIP:DMG: <RE> Sending report (payload has 907 bytes)..._

At 426: `err = SendReport(apReadHandler, std::move(bufHandle), hasMoreChunks);`

Enter `Engine::SendReport(…)`
At 638: `err = apReadHandler->SendReportData(std::move(aPayload), aHasMoreChunks);`

Enter `ReadHandler::SendReportData(…)`

_CHIP:DMG: IM RH moving to [AwaitingReportResponse]_

At 221: `mpExchangeCtx->SendMessage(Protocols::InteractionModel::MsgType::ReportData, std::move(aPayload),`
At 222: `                           Messaging::SendFlags(noResponseExpected ? Messaging::SendMessageFlags::kNone`
At 223: `                                                                   : Messaging::SendMessageFlags::kExpectResponse));`

Enter `SendMessage(…)`
At 126: `return SendMessage(Protocols::MessageTypeTraits<MessageType>::ProtocolId(), to_underlying(msgType), std::move(msgPayload), …)`

Enter `ExchangeContext::SendMessage(…)`
At 179: `CHIP_ERROR err = mDispatch.SendMessage(GetExchangeMgr()->GetSessionManager(), mSession.Get(), mExchangeId, IsInitiator(),`
At 180: `                                       GetReliableMessageContext(), reliableTransmissionRequested, protocolId, msgType,`
At 181: `                                       std::move(msgBuf));`

Enter `ExchangeMessageDispatch::SendMessage(…)`

_CHIP:EM: Piggybacking Ack for MessageCounter:12451190 on exchange: 22567r_

___Here___ the `RetransTableEntry` for MessageCounter:__10983592__ is created and retained in `entryOwner`:

At 80: `ReturnErrorOnFailure(reliableMessageMgr->AddToRetransTable(reliableMessageContext, &entry));`
At 84: `std::unique_ptr<ReliableMessageMgr::RetransTableEntry, decltype(deleter)> entryOwner(entry, deleter);`

_CHIP:IN: Prepared secure message 0x7ffff7fb4b88 to 0x000000000001E306 (1)  of type 0x5 and protocolId (0, 1) on exchange 22567r with MessageCounter:__10983592__._

At 87: `CHIP_ERROR err = sessionManager->SendPreparedMessage(session, entryOwner->retainedBuf);`

Enter `SessionManager::SendPreparedMessage(…)`

_CHIP:IN: Sending encrypted msg 0x7ffff7fb4b88 with MessageCounter:__10983592__ to 0x000000000001E306 (1) at monotonic time: 1485439570 msec_

At 289: `return mTransportMgr->SendMessage(*destination, std::move(msgBuf));`

Enter `Transport::Tuple::SendMessage()`

Enter `LoopbackTransport::SendMessage()`
At 112: `HandleMessageReceived(address, std::move(receivedMessage));`

Enter `Transport::Base::HandleMessageReceived()`

Enter `TransportMgrBase::HandleMessageReceived(…)`
At 71: `mSessionManager->OnMessageReceived(peerAddress, std::move(msg));`

Enter `SessionManager::OnMessageReceived(…)`
At 399: `SecureUnicastMessageDispatch(packetHeader, peerAddress, std::move(msg));`

Enter `SessionManager::SecureUnicastMessageDispatch(…)`
At 510: `if (SecureMessageCodec::Decrypt(secureSession->GetCryptoContext(), payloadHeader, packetHeader, msg) != CHIP_NO_ERROR)`

[... 129 lines cut ...]
Return to `ExchangeContext::SendMessage()`
Return to `SendMessage()`
Return to `StatusResponse::Send()`
Return to `ReadClient::ProcessReportData()`
Return to `ReadClient::OnMessageReceived()`
Return to `ExchangeContext::HandleMessage()`
Return to `ExchangeManager::OnMessageReceived()`
Return to `SessionManager::SecureUnicastMessageDispatch()`
Return to `SessionManager::OnMessageReceived()`
Return to `TransportMgrBase::HandleMessageReceived()`
Return to `Transport::Base::HandleMessageReceived()`
Return to `LoopbackTransport::SendMessage()`
Return to `Transport::Tuple::SendMessage()`
Return to `SessionManager::SendPreparedMessage()`
Return to `ExchangeMessageDispatch::SendMessage()`

___Here___ we are back in `ExchangeMessageDispatch::SendMessage(…)` with `entryOwner` pointing to the deleted `RetransTableEntry`.

At 102: `reliableMessageMgr->StartRetransmision(entryOwner.release());`

Enter `ReliableMessageMgr::StartRetransmision(…)`

___Here___ we have read and write of the deleted `RetransTableEntry`:

At 207: `entry->nextRetransTime = System::SystemClock().GetMonotonicTimestamp() + entry->ec->GetSessionHandle()->GetMRPConfig().mIdleRetransTimeout;`

bzbarsky-apple, 2022-02-04:
So the fundamental problem, I expect, is that this test uses the sync-message-delivery loopback mode that I thought we had stopped using....  but it looks like a bunch of tests still use it:

* TestWriteInteraction
* TestCommandInteraction
* TestEventLogging
* TestReportingEngine
* TestReadInteraction
* TestInteractionModelEngine.cpp
* TestBufferedReadCallback
* TestAttributeCache

We should probably fix them all to not do that and see whether that addresses this issue.

@mrjerryjohns

kpschoedel, 2022-02-04:
Suspected it might be test-only when I hit `LoopbackTransport`.

The Interaction tests and TestReportingEngine hit this; the others pass ASAN.

bzbarsky-apple, 2022-06-28:
Looks like this is fixed now that all the unit tests use async messaging.  I just tried making the retrans table into an ObjectPool and it passes for me.  See https://github.com/project-chip/connectedhomeip/pull/20058

closed 2022-06-28 by bzbarsky-apple