project-chip/connectedhomeip #14462: [TC-BR-3] missing `node-label` attribute in `bridgeddevicebasic` chip-tool cluster for TE7.5

liamgonyea, 2022-01-27:
#### Problem
When running TC-BR-3, Step 1a and 1c, the `node-label` attribute for the `bridgeddevicebasic` cluster is needed to check the node label and if it is changed in Step 1b. 
[image]

#### Test Case
3.1.3. [TC-BR-3] Changing name and state of a bridged device
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/bridge.adoc#313-tc-br-3-changing-name-and-state-of-a-bridged-device

#### Logs
See line: `[1643250202041] [30359:6757409] CHIP: [TOO] Unknown attribute: node-label`
[consolidated_mac_accessory_TC-BR-3.txt](https://github.com/project-chip/connectedhomeip/files/7954106/consolidated_mac_accessory_TC-BR-3.txt)

turon, 2022-02-01:
Passing cert tests of bridged devices is required for v1.
Removing tag `v1_triage_split_4`

bzbarsky-apple, 2022-05-05:
@liamgonyea is this a bug report for chip-tool or for the server side?