project-chip/connectedhomeip #14472: TE7.5 [TC-GC-1.1] VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0

nanmu2018, 2022-01-28:
**Problem：**
[TC-GC-1.1] VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted

**platform:**
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba

```
ubuntu@matter-7-5:~/connectedhomeip/out/standalone$ sudo ./chip-tool generalcommissioning read breadcrumb 12344321 1
[1643274770.425696][5359:5359] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-1yicuE)
[1643274770.426351][5359:5359] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643274770.426436][5359:5359] CHIP:DL: NVS set: chip-counters/reboot-count = 7 (0x7)
[1643274770.427365][5359:5359] CHIP:DL: Got Ethernet interface: eth0
[1643274770.427966][5359:5359] CHIP:DL: Found the primary Ethernet interface:eth0
[1643274770.428687][5359:5359] CHIP:DL: Got WiFi interface: wlan0
[1643274770.428761][5359:5359] CHIP:DL: Failed to reset WiFi statistic counts
[1643274770.428830][5359:5359] CHIP:IN: UDP::Init bind&listen port=5542
[1643274770.428942][5359:5359] CHIP:IN: UDP::Init bound to port=5542
[1643274770.428970][5359:5359] CHIP:IN: UDP::Init bind&listen port=5542
[1643274770.429064][5359:5359] CHIP:IN: UDP::Init bound to port=5542
[1643274770.429089][5359:5359] CHIP:IN: TransportMgr initialized
[1643274770.429128][5359:5359] CHIP:DIS: Init fabric pairing table with server storage
[1643274770.429415][5359:5359] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-cfYBjG)
[1643274770.430061][5359:5359] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643274770.430116][5359:5359] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 6000 (0x1770)
[1643274770.430239][5359:5359] CHIP:ZCL: Using ZAP configuration...
[1643274770.433501][5359:5359] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643274770.440824][5359:5359] CHIP:DL: MDNS failed to join multicast group on veth7bb0b90 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643274770.442726][5359:5359] CHIP:DL: MDNS failed to join multicast group on vethefb2194 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643274770.444557][5359:5359] CHIP:DL: MDNS failed to join multicast group on veth9fd920d for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643274770.446326][5359:5359] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643274770.446866][5359:5359] CHIP:CTL: System State Initialized...
[1643274770.449544][5359:5359] CHIP:CTL: Generating NOC
[1643274770.449858][5359:5359] CHIP:CTL: Generating ICAC
[1643274770.450775][5359:5359] CHIP:DIS: Verifying the received credentials
[1643274770.452529][5359:5359] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1643274770.452565][5359:5359] CHIP:IN: 0x04, 0x35, 0xbc, 0x79, 0x0d, 0x81, 0xba, 0x7b,
[1643274770.452590][5359:5359] CHIP:IN: 0x01, 0xfb, 0xb5, 0x3e, 0x39, 0x88, 0x07, 0x21,
[1643274770.452614][5359:5359] CHIP:IN: 0x24, 0x9b, 0xc9, 0xbc, 0x44, 0x0d, 0x31, 0x0a,
[1643274770.452637][5359:5359] CHIP:IN: 0x74, 0x4c, 0xdc, 0x4e, 0x18, 0xb3, 0x14, 0xf5,
[1643274770.452660][5359:5359] CHIP:IN: 0xf4, 0xd4, 0x92, 0x47, 0x5c, 0x88, 0x4b, 0x9d,
[1643274770.452682][5359:5359] CHIP:IN: 0xdc, 0x65, 0x2b, 0xe2, 0xc5, 0xcc, 0xc1, 0x84,
[1643274770.452705][5359:5359] CHIP:IN: 0xe4, 0x4f, 0x08, 0xf7, 0x18, 0x85, 0x73, 0x7c,
[1643274770.452728][5359:5359] CHIP:IN: 0x15, 0x0b, 0xfd, 0xd7, 0x01, 0x88, 0x97, 0x2d,
[1643274770.452748][5359:5359] CHIP:IN: 0xde,
[1643274770.452801][5359:5359] CHIP:IN: Generated compressed fabric ID
[1643274770.452827][5359:5359] CHIP:IN: 0x7b, 0x18, 0xf4, 0x57, 0xe9, 0x2a, 0xda, 0xfd,
[1643274770.452855][5359:5359] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1643274770.452878][5359:5359] CHIP:DIS: Assigned compressed fabric ID: 0x7B18F457E92ADAFD, node ID: 0x000000000001B669
[1643274770.452901][5359:5359] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x7B18F457E92ADAFD
[1643274770.452968][5359:5359] CHIP:IN: UDP::Init bind&listen port=5550
[1643274770.453105][5359:5359] CHIP:IN: UDP::Init bound to port=5550
[1643274770.453131][5359:5359] CHIP:IN: UDP::Init bind&listen port=5550
[1643274770.453221][5359:5359] CHIP:IN: UDP::Init bound to port=5550
[1643274770.453246][5359:5359] CHIP:IN: TransportMgr initialized
[1643274770.454079][5359:5359] CHIP:CTL: Generating NOC
[1643274770.454368][5359:5359] CHIP:CTL: Generating ICAC
[1643274770.455144][5359:5359] CHIP:DIS: Verifying the received credentials
[1643274770.456855][5359:5359] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1643274770.456891][5359:5359] CHIP:IN: 0x04, 0x35, 0xbc, 0x79, 0x0d, 0x81, 0xba, 0x7b,
[1643274770.456915][5359:5359] CHIP:IN: 0x01, 0xfb, 0xb5, 0x3e, 0x39, 0x88, 0x07, 0x21,
[1643274770.456939][5359:5359] CHIP:IN: 0x24, 0x9b, 0xc9, 0xbc, 0x44, 0x0d, 0x31, 0x0a,
[1643274770.456962][5359:5359] CHIP:IN: 0x74, 0x4c, 0xdc, 0x4e, 0x18, 0xb3, 0x14, 0xf5,
[1643274770.456984][5359:5359] CHIP:IN: 0xf4, 0xd4, 0x92, 0x47, 0x5c, 0x88, 0x4b, 0x9d,
[1643274770.457007][5359:5359] CHIP:IN: 0xdc, 0x65, 0x2b, 0xe2, 0xc5, 0xcc, 0xc1, 0x84,
[1643274770.457030][5359:5359] CHIP:IN: 0xe4, 0x4f, 0x08, 0xf7, 0x18, 0x85, 0x73, 0x7c,
[1643274770.457053][5359:5359] CHIP:IN: 0x15, 0x0b, 0xfd, 0xd7, 0x01, 0x88, 0x97, 0x2d,
[... 148 lines cut ...]
[1643274773.137051][5359:5364] CHIP:DMG:                                }
[1643274773.137178][5359:5364] CHIP:DMG:
[1643274773.137293][5359:5364] CHIP:DMG:                                StatusIB =
[1643274773.137372][5359:5364] CHIP:DMG:                                {
[1643274773.137487][5359:5364] CHIP:DMG:                                        status = 0x86,
[1643274773.137610][5359:5364] CHIP:DMG:                                },
[1643274773.138218][5359:5364] CHIP:DMG:
[1643274773.138309][5359:5364] CHIP:DMG:                        },
[1643274773.138442][5359:5364] CHIP:DMG:
[1643274773.138523][5359:5364] CHIP:DMG:                },
[1643274773.138609][5359:5364] CHIP:DMG:
[1643274773.138670][5359:5364] CHIP:DMG:        ],
[1643274773.138747][5359:5364] CHIP:DMG:
[1643274773.138807][5359:5364] CHIP:DMG:        SuppressResponse = true,
[1643274773.138863][5359:5364] CHIP:DMG: }
[1643274773.139131][5359:5364] CHIP:TOO: Default Failure Response: 0x86
[1643274773.139282][5359:5364] CHIP:EM: Sending Standalone Ack for MessageCounter:16127354 on exchange 18267i
[1643274773.139427][5359:5364] CHIP:IN: Prepared encrypted message 0xffffa37ed9d8 to 0x0000000000BC5C01 (1)  of type 0x10 and protocolId (0, 0) on exchange 18267i with MessageCounter:6773994.
[1643274773.139522][5359:5364] CHIP:IN: Sending encrypted msg 0xffffa37ed9d8 with MessageCounter:6773994 to 0x0000000000BC5C01 (1) at monotonic time: 22087161 msec
[1643274773.139776][5359:5364] CHIP:EM: Flushed pending ack for MessageCounter:16127354 on exchange 18267i
[1643274773.140159][5359:5359] CHIP:-: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1643274773.140254][5359:5359] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error
[1643274773.147093][5359:5359] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted
```

andy31415, 2022-01-31:
@nanmu2018 - please follow up with test cases that fail inside this issue instead of creating separate issues for every case. I imagine whoever will end up debugging and fixing this will want to validate all paths as part of a single issue instead of multiple ones.

jmeg-sfy, 2022-02-03:
Bad endpoint is selected on this logs 

I guess we can close this issue

bzbarsky-apple, 2022-05-05:
Resolving: The read is using the wrong endpoint, and on tip (00ccf441a52976b37b60b5219a0ad3e2bd612bc1) we don't crash in this error scenario.

justinsignify494, 2024-08-26:
Having this same problem!
Any solution as of now?