project-chip/connectedhomeip #14475: TE7.5 [TC-SU] VerifyOrDie failure at ../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0 Aborted (core dumped)

nanmu2018, 2022-01-28:
Problem：
[TC-SU] Problem：VerifyOrDie failure at ../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)

platform:
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba

```
ubuntu@matter-7-5:~/connectedhomeip$ ./out/chip-tool pairing onnetwork 1 73141520
[1643267920.768857][4895:4895] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-1hc9Pz)
[1643267920.769527][4895:4895] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643267920.769583][4895:4895] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1643267920.770630][4895:4895] CHIP:DL: Got Ethernet interface: eth0
[1643267920.771329][4895:4895] CHIP:DL: Found the primary Ethernet interface:eth0
[1643267920.772055][4895:4895] CHIP:DL: Got WiFi interface: wlan0
[1643267920.772141][4895:4895] CHIP:DL: Failed to reset WiFi statistic counts
[1643267920.772228][4895:4895] CHIP:IN: UDP::Init bind&listen port=5542
[1643267920.772399][4895:4895] CHIP:IN: UDP::Init bound to port=5542
[1643267920.772431][4895:4895] CHIP:IN: UDP::Init bind&listen port=5542
[1643267920.772531][4895:4895] CHIP:IN: UDP::Init bound to port=5542
[1643267920.772559][4895:4895] CHIP:IN: TransportMgr initialized
[1643267920.772603][4895:4895] CHIP:DIS: Init fabric pairing table with server storage
[1643267920.772958][4895:4895] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-YciZx7)
[1643267920.777642][4895:4895] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643267920.777710][4895:4895] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 3000 (0xBB8)
[1643267920.777895][4895:4895] CHIP:ZCL: Using ZAP configuration...
[1643267920.781408][4895:4895] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643267920.791760][4895:4895] CHIP:DL: MDNS failed to join multicast group on veth7bb0b90 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643267920.794032][4895:4895] CHIP:DL: MDNS failed to join multicast group on vethefb2194 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643267920.796070][4895:4895] CHIP:DL: MDNS failed to join multicast group on veth9fd920d for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643267920.797911][4895:4895] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643267920.798227][4895:4895] CHIP:CTL: System State Initialized...
[1643267920.801004][4895:4895] CHIP:CTL: Generating NOC
[1643267920.801390][4895:4895] CHIP:CTL: Generating ICAC
[1643267920.802223][4895:4895] CHIP:DIS: Verifying the received credentials
[1643267920.804322][4895:4895] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1643267920.804380][4895:4895] CHIP:IN: 0x04, 0x35, 0xbc, 0x79, 0x0d, 0x81, 0xba, 0x7b,
[1643267920.804403][4895:4895] CHIP:IN: 0x01, 0xfb, 0xb5, 0x3e, 0x39, 0x88, 0x07, 0x21,
[1643267920.804426][4895:4895] CHIP:IN: 0x24, 0x9b, 0xc9, 0xbc, 0x44, 0x0d, 0x31, 0x0a,
[1643267920.804450][4895:4895] CHIP:IN: 0x74, 0x4c, 0xdc, 0x4e, 0x18, 0xb3, 0x14, 0xf5,
[1643267920.804473][4895:4895] CHIP:IN: 0xf4, 0xd4, 0x92, 0x47, 0x5c, 0x88, 0x4b, 0x9d,
[1643267920.804496][4895:4895] CHIP:IN: 0xdc, 0x65, 0x2b, 0xe2, 0xc5, 0xcc, 0xc1, 0x84,
[1643267920.804519][4895:4895] CHIP:IN: 0xe4, 0x4f, 0x08, 0xf7, 0x18, 0x85, 0x73, 0x7c,
[1643267920.804542][4895:4895] CHIP:IN: 0x15, 0x0b, 0xfd, 0xd7, 0x01, 0x88, 0x97, 0x2d,
[1643267920.804562][4895:4895] CHIP:IN: 0xde,
[1643267920.804619][4895:4895] CHIP:IN: Generated compressed fabric ID
[1643267920.804646][4895:4895] CHIP:IN: 0x7b, 0x18, 0xf4, 0x57, 0xe9, 0x2a, 0xda, 0xfd,
[1643267920.804679][4895:4895] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1643267920.804703][4895:4895] CHIP:DIS: Assigned compressed fabric ID: 0x7B18F457E92ADAFD, node ID: 0x000000000001B669
[1643267920.804726][4895:4895] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x7B18F457E92ADAFD
[1643267920.804808][4895:4895] CHIP:IN: UDP::Init bind&listen port=5550
[1643267920.804942][4895:4895] CHIP:IN: UDP::Init bound to port=5550
[1643267920.804971][4895:4895] CHIP:IN: UDP::Init bind&listen port=5550
[1643267920.805064][4895:4895] CHIP:IN: UDP::Init bound to port=5550
[1643267920.805089][4895:4895] CHIP:IN: TransportMgr initialized
[1643267920.806001][4895:4895] CHIP:CTL: Generating NOC
[1643267920.806329][4895:4895] CHIP:CTL: Generating ICAC
[1643267920.807348][4895:4895] CHIP:DIS: Verifying the received credentials
[1643267920.809519][4895:4895] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1643267920.809573][4895:4895] CHIP:IN: 0x04, 0x35, 0xbc, 0x79, 0x0d, 0x81, 0xba, 0x7b,
[1643267920.809596][4895:4895] CHIP:IN: 0x01, 0xfb, 0xb5, 0x3e, 0x39, 0x88, 0x07, 0x21,
[1643267920.809620][4895:4895] CHIP:IN: 0x24, 0x9b, 0xc9, 0xbc, 0x44, 0x0d, 0x31, 0x0a,
[1643267920.809643][4895:4895] CHIP:IN: 0x74, 0x4c, 0xdc, 0x4e, 0x18, 0xb3, 0x14, 0xf5,
[1643267920.809665][4895:4895] CHIP:IN: 0xf4, 0xd4, 0x92, 0x47, 0x5c, 0x88, 0x4b, 0x9d,
[1643267920.809689][4895:4895] CHIP:IN: 0xdc, 0x65, 0x2b, 0xe2, 0xc5, 0xcc, 0xc1, 0x84,
[1643267920.809711][4895:4895] CHIP:IN: 0xe4, 0x4f, 0x08, 0xf7, 0x18, 0x85, 0x73, 0x7c,
[1643267920.809734][4895:4895] CHIP:IN: 0x15, 0x0b, 0xfd, 0xd7, 0x01, 0x88, 0x97, 0x2d,
[... 26 lines cut ...]
[1643267920.814579][4895:4895] CHIP:DIS: Added new fabric at index: 3, Initialized: 1
[1643267920.814602][4895:4895] CHIP:DIS: Assigned compressed fabric ID: 0x7082E83747295B21, node ID: 0x000000000001B669
[1643267920.814624][4895:4895] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0x7082E83747295B21
[1643267920.814708][4895:4895] CHIP:IN: UDP::Init bind&listen port=5550
[1643267920.814839][4895:4895] CHIP:IN: UDP::Init bound to port=5550
[1643267920.814866][4895:4895] CHIP:IN: UDP::Init bind&listen port=5550
[1643267920.814955][4895:4895] CHIP:IN: UDP::Init bound to port=5550
[1643267920.814979][4895:4895] CHIP:IN: TransportMgr initialized
[1643267920.815278][4895:4901] CHIP:DL: CHIP task running
[1643267920.816016][4895:4901] CHIP:DIS: Attempt to mDNS broadcast failed:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1643267920.816361][4895:4901] CHIP:DIS: Attempt to mDNS broadcast failed:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1643267920.816779][4895:4901] CHIP:DIS: Attempt to mDNS broadcast failed:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1643267920.816938][4895:4901] CHIP:DIS: Attempt to mDNS broadcast failed:  ../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
[1643267920.817755][4895:4901] CHIP:TOO: Discovered Device: 2001:db8::849c:7f62:c927:b009:5540
[1643267920.817928][4895:4901] CHIP:SC: Assigned local session key ID 1
[1643267920.818080][4895:4901] CHIP:SC: Including MRP parameters in PBKDF param request
[1643267920.818147][4895:4901] CHIP:IN: Prepared plaintext message 0xaaaaeb3ff460 to 0x0000000000000000 (0)  of type 0x20 and protocolId (0, 0) on exchange 64037i with MessageCounter:1965855423.
[1643267920.818189][4895:4901] CHIP:IN: Sending plaintext msg 0xaaaaeb3ff460 with MessageCounter:1965855423 to 0x0000000000000000 at monotonic time: 15234839 msec
[1643267920.818350][4895:4901] CHIP:SC: Sent PBKDF param request
[1643267960.821645][4895:4901] CHIP:TOO: Pairing Failure: ../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1553: CHIP Error 0x00000032: Timeout
[... cut ...] iceController.cpp:1553: CHIP Error 0x00000032: Timeout at ../examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1643267960.822301][4895:4895] CHIP:TOO: Run command failure: ../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1553: CHIP Error 0x00000032: Timeout
[1643267960.828738][4895:4895] CHIP:SPT: VerifyOrDie failure at ../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)
```

andy31415, 2022-01-30:
Seems like a duplicate of #14476 but with a different test case run.

andy31415, 2022-01-30:
Actually will use #14472 as the duplicate as it was the first ever reported. Will close all the other duplicates and keep the 14472 open.

closed 2022-01-30 by andy31415