project-chip/connectedhomeip #14482: TE7.5 [TC-IDM-5.2]CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0 Aborted (core dumped)

nanmu2018, 2022-01-28:
Problem：
[TC-IDM-5.2] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)

platform:
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba

```
ubuntu@matter-7-5:~/connectedhomeip/out/standalone$ ./chip-tool basic  mfg-specific-ping  12344321 1   --commissioner-name  alpha  --timedInteractionTimeoutMs   -200
[1643351007.436518][3699:3699] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-utN8OD)
[1643351007.437242][3699:3699] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643351007.437301][3699:3699] CHIP:DL: NVS set: chip-counters/reboot-count = 7 (0x7)
[1643351007.438267][3699:3699] CHIP:DL: Got Ethernet interface: eth0
[1643351007.439021][3699:3699] CHIP:DL: Found the primary Ethernet interface:eth0
[1643351007.439746][3699:3699] CHIP:DL: Got WiFi interface: wlan0
[1643351007.439826][3699:3699] CHIP:DL: Failed to reset WiFi statistic counts
[1643351007.439903][3699:3699] CHIP:IN: UDP::Init bind&listen port=5542
[1643351007.440028][3699:3699] CHIP:IN: UDP::Init bound to port=5542
[1643351007.440059][3699:3699] CHIP:IN: UDP::Init bind&listen port=5542
[1643351007.440164][3699:3699] CHIP:IN: UDP::Init bound to port=5542
[1643351007.440194][3699:3699] CHIP:IN: TransportMgr initialized
[1643351007.440237][3699:3699] CHIP:DIS: Init fabric pairing table with server storage
[1643351007.440542][3699:3699] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-Udqb7e)
[1643351007.441171][3699:3699] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1643351007.441229][3699:3699] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 7000 (0x1B58)
[1643351007.441361][3699:3699] CHIP:ZCL: Using ZAP configuration...
[1643351007.444842][3699:3699] CHIP:DL: MDNS failed to join multicast group on wlan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643351007.452390][3699:3699] CHIP:DL: MDNS failed to join multicast group on vethda6fe02 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643351007.454312][3699:3699] CHIP:DL: MDNS failed to join multicast group on vethd1a29ba for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643351007.456148][3699:3699] CHIP:DL: MDNS failed to join multicast group on veth6fb6734 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643351007.458102][3699:3699] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
[1643351007.458448][3699:3699] CHIP:CTL: System State Initialized...
[1643351007.461157][3699:3699] CHIP:CTL: Generating NOC
[1643351007.461658][3699:3699] CHIP:CTL: Generating ICAC
[1643351007.462457][3699:3699] CHIP:DIS: Verifying the received credentials
[1643351007.464212][3699:3699] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1643351007.464250][3699:3699] CHIP:IN: 0x04, 0x25, 0xeb, 0x71, 0x73, 0xbf, 0x52, 0x2a,
[1643351007.464275][3699:3699] CHIP:IN: 0xda, 0x7a, 0x50, 0x2c, 0x64, 0x98, 0x7c, 0x8a,
[1643351007.464298][3699:3699] CHIP:IN: 0xb4, 0x3a, 0xe0, 0x4b, 0xbf, 0xcc, 0xed, 0xe2,
[1643351007.464321][3699:3699] CHIP:IN: 0x28, 0xef, 0x87, 0xab, 0xcf, 0x10, 0xc4, 0x45,
[1643351007.464344][3699:3699] CHIP:IN: 0x5d, 0x50, 0xbf, 0xed, 0xff, 0x62, 0xe9, 0x87,
[1643351007.464366][3699:3699] CHIP:IN: 0xf6, 0xac, 0x90, 0x2d, 0x2c, 0x49, 0x6d, 0x6d,
[1643351007.464388][3699:3699] CHIP:IN: 0x93, 0x7e, 0xc6, 0xb4, 0x18, 0x65, 0x90, 0xe2,
[1643351007.464411][3699:3699] CHIP:IN: 0x2c, 0x80, 0xe1, 0x25, 0x5c, 0xb7, 0xfc, 0x0e,
[1643351007.464431][3699:3699] CHIP:IN: 0x3d,
[1643351007.464485][3699:3699] CHIP:IN: Generated compressed fabric ID
[1643351007.464511][3699:3699] CHIP:IN: 0x56, 0x97, 0xd4, 0x7a, 0x18, 0xde, 0x2e, 0x34,
[1643351007.464542][3699:3699] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
[1643351007.464564][3699:3699] CHIP:DIS: Assigned compressed fabric ID: 0x5697D47A18DE2E34, node ID: 0x000000000001B669
[1643351007.464587][3699:3699] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x5697D47A18DE2E34
[1643351007.464654][3699:3699] CHIP:IN: UDP::Init bind&listen port=5550
[1643351007.464804][3699:3699] CHIP:IN: UDP::Init bound to port=5550
[1643351007.464831][3699:3699] CHIP:IN: UDP::Init bind&listen port=5550
[1643351007.464922][3699:3699] CHIP:IN: UDP::Init bound to port=5550
[1643351007.464946][3699:3699] CHIP:IN: TransportMgr initialized
[1643351007.466117][3699:3699] CHIP:CTL: Generating NOC
[1643351007.466433][3699:3699] CHIP:CTL: Generating ICAC
[1643351007.467151][3699:3699] CHIP:DIS: Verifying the received credentials
[1643351007.468880][3699:3699] CHIP:IN: Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1643351007.468919][3699:3699] CHIP:IN: 0x04, 0x25, 0xeb, 0x71, 0x73, 0xbf, 0x52, 0x2a,
[1643351007.468943][3699:3699] CHIP:IN: 0xda, 0x7a, 0x50, 0x2c, 0x64, 0x98, 0x7c, 0x8a,
[1643351007.468967][3699:3699] CHIP:IN: 0xb4, 0x3a, 0xe0, 0x4b, 0xbf, 0xcc, 0xed, 0xe2,
[1643351007.468989][3699:3699] CHIP:IN: 0x28, 0xef, 0x87, 0xab, 0xcf, 0x10, 0xc4, 0x45,
[1643351007.469012][3699:3699] CHIP:IN: 0x5d, 0x50, 0xbf, 0xed, 0xff, 0x62, 0xe9, 0x87,
[1643351007.469035][3699:3699] CHIP:IN: 0xf6, 0xac, 0x90, 0x2d, 0x2c, 0x49, 0x6d, 0x6d,
[1643351007.469057][3699:3699] CHIP:IN: 0x93, 0x7e, 0xc6, 0xb4, 0x18, 0x65, 0x90, 0xe2,
[1643351007.469079][3699:3699] CHIP:IN: 0x2c, 0x80, 0xe1, 0x25, 0x5c, 0xb7, 0xfc, 0x0e,
[... 165 lines cut ...]
[1643351010.035763][3699:3705] CHIP:DMG:
[1643351010.035803][3699:3705] CHIP:DMG:                                StatusIB =
[1643351010.035846][3699:3705] CHIP:DMG:                                {
[1643351010.035888][3699:3705] CHIP:DMG:                                        status = 0x85,
[1643351010.035931][3699:3705] CHIP:DMG:                                },
[1643351010.035975][3699:3705] CHIP:DMG:
[1643351010.036012][3699:3705] CHIP:DMG:                        },
[1643351010.036059][3699:3705] CHIP:DMG:
[1643351010.036094][3699:3705] CHIP:DMG:                },
[1643351010.036139][3699:3705] CHIP:DMG:
[1643351010.036172][3699:3705] CHIP:DMG:        ],
[1643351010.036215][3699:3705] CHIP:DMG:
[1643351010.036246][3699:3705] CHIP:DMG: },
[1643351010.036336][3699:3705] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0028 Command=0x1002_0000 Status=0x85
[1643351010.036402][3699:3705] CHIP:TOO: Default Failure Response: 0x85
[1643351010.036485][3699:3705] CHIP:DMG: ICR moving to [AwaitingDe]
[1643351010.036539][3699:3705] CHIP:EM: Sending Standalone Ack for MessageCounter:14547385 on exchange 16647i
[1643351010.036647][3699:3705] CHIP:IN: Prepared encrypted message 0xffff8095e9d8 to 0x0000000000BC5C01 (1)  of type 0x10 and protocolId (0, 0) on exchange 16647i with MessageCounter:4818780.
[1643351010.036714][3699:3705] CHIP:IN: Sending encrypted msg 0xffff8095e9d8 with MessageCounter:4818780 to 0x0000000000BC5C01 (1) at monotonic time: 4208679 msec
[1643351010.036911][3699:3705] CHIP:EM: Flushed pending ack for MessageCounter:14547385 on exchange 16647i
[1643351010.037332][3699:3699] CHIP:-: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error at ../../examples/chip-tool/commands/common/CHIPCommand.cpp:59
[1643351010.037442][3699:3699] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/zzz_generated/chip-tool/zap-generated/cluster/Commands.h:2502: CHIP Error 0x000000AC: Internal error
[1643351010.042609][3699:3699] CHIP:SPT: VerifyOrDie failure at ../../examples/chip-tool/third_party/connectedhomeip/src/inet/InetLayer.h:121: sEndPointPool.Allocated() == 0
Aborted (core dumped)
```

andy31415, 2022-01-30:
What were the steps involved here?

andy31415, 2022-01-30:
Seems like a duplicate of #14472 but with a different test case run.
We seem to have quite a few stack traces showing potential memory corruption.

closed 2022-01-30 by andy31415