project-chip/connectedhomeip #14508: Revisit client side cluster related code

erjiaqing, 2022-01-28:
One more thing: this part does not look right to me.... Why are we generating things for client sides of clusters here?

Let's leave this as-is for now, but do a followup to not generate this... and probably not generate ClusterRevision for client clusters either, since it's not observable...

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/14299#discussion_r794022110_

ZAP will generate meta for client cluster, but the client should not have cluster data per spec, should revisit related code in code base.

bzbarsky-apple, 2022-01-29:
We need _some_ metadata for client clusters, because Descriptor Cluster needs to be able to show info about them existing, right?

What we should not be generating are:

1. ClusterRevision attributes for client clusters, with space in the attr store and all that.
2. CommandList metadata bits for client clusters.

And I guess we could try to handle the Descriptor bits via some mechanism other than EmberAfCluster entries, but that will take a bit more surgery, whereas not generating the above two is an easy space win.

bzbarsky-apple, 2022-02-21:
Fixed in  #14677

closed 2022-02-21 by bzbarsky-apple