project-chip/connectedhomeip #14512: Possible memory corruption in GroupSessionTable

kpschoedel, 2022-01-28:
#### Problem

If `GroupSessionTable::mEntries` uses a heap object pool, `TestWriteInteraction` segfaults:
```
  ⋮
[1643341603.017436][32137:32137] CHIP:DMG: Received group attribute write for Group=517 Cluster=0x0000_0003 attribute=0x0000_0004
277
[1643341603.017446][32137:32137] CHIP:DMG: IM WH moving to [Uninitialized]
278
Segmentation fault (core dumped)
279
INF Test 1/1: [FAIL] TestWriteInteraction
```

Memory corruption may be masked by the use of an inline BitMapObjectPool.

(Observed in multiple workflows in [fork CI](https://github.com/kpschoedel/connectedhomeip/runs/4974007011); I don't have a local reproduction yet.)

 #### Proposed Solution

Investigate and resolve.

kpschoedel, 2022-02-07:
```
644  Optional<SessionHandle> session = CreateGroupSession(groupContext.group_id, groupContext.fabric_index);
645
646  VerifyOrReturn(session.HasValue(), ChipLogError(Inet, "Error when creating group session handle."));
647  Transport::GroupSession * groupSession = session.Value()->AsGroupSession();
648
649  CHIP_TRACE_MESSAGE_RECEIVED(payloadHeader, packetHeader, groupSession, peerAddress, msg->Start(), msg->TotalLength());
650  mCB->OnMessageReceived(packetHeader, payloadHeader, session.Value(), peerAddress, isDuplicate, std::move(msg));
651
652  RemoveGroupSession(groupSession);
```

This is similar to #14444. `Sessionhandle` uses `ReferenceCountedHandle` but `GroupSession` doesn't have a reference count; it inherits from `Session`:

``` 
62     virtual void Retain() {}
63     virtual void Release() {}
```

bzbarsky-apple, 2022-02-08:
Group sessions should generally be stack-allocated to start with, I suspect.  Have an item to do that on my whiteboard right now....

kghost, 2022-02-09:
I'll make the group session entry ephemeral.

bzbarsky-apple, 2022-06-28:
Group sessions are stack-only now; there is no mEntries table.

closed 2022-06-28 by bzbarsky-apple