project-chip/connectedhomeip #14521: [TC_MC_4.1] Mac Address returned is all 0’s for `wake-on-lan-mac-address` attribute for TE7.5

liamgonyea, 2022-01-28:
#### Problem
When running [TC_MC_4.1], Step 1, the Mac Address returned is all 0’s for `wake-on-lan-mac-address` attribute for the `wakeonlan ` cluster.
Expect to see the mac address of the device, in this case, the RasPi device running chip-tv-app.

[image]

#### Test Case
3.6.1. [TC-MC-4.1] Wake-On LAN Verification
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#361-tc-mc-41-wake-on-lan-verification

#### Logs
TE7.5 commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`

`./out/debug/chip-tool wakeonlan read wake-on-lan-mac-address 1234 1`
> [1643072172281] [18045:5782860] CHIP: [TOO] WakeOnLan.WakeOnLanMacAddress response: 00:00:00:00:00

[consolidated_mac_accessory_TC-MC-4-1.txt](https://github.com/project-chip/connectedhomeip/files/7960653/consolidated_mac_accessory_TC-MC-4-1.txt)

lazarkov, 2022-02-10:
Please use to build the chip tv app:
`./gn_build.sh chip_config_network_layer_ble=false chip_enable_wake_on_lan=true `

lazarkov, 2022-02-10:
or at least add the flag `chip_enable_wake_on_lan=true`