project-chip/connectedhomeip #14530: Followup: we should decide what we do if we have a chunked result come in and aVersion does not match mDataVersion.

yunhanw-google, 2022-01-28:
Followup: we should decide what we do if we have a chunked result come in and aVersion does not match mDataVersion.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/13400#discussion_r794776676_

bzbarsky-apple, 2022-01-28:
This is specifically for what happens for a chunked read on the client side.

mrjerryjohns, 2022-02-10:
Note that this can happen if a mid-way through processing a chunked read, the data version on the server changed and it had to 'reset' and create a new stream of AttributeDataIBs. This can happen at the boundary of an attribute, or intra-attribute (in the case of a list). In the latter, it should always result in a replace operation commencing the new 'reseted' stream.

The only illegal case here is if mid-way through the stream of IBs for a chunked list, we got a non-replace operation with a different data version. Not sure what we should do here...