project-chip/connectedhomeip #14542: Allow UDP listening port selection for Android

g-coppock, 2022-01-28:
#### Problem

In `AndroidDeviceControllerWrapper.cpp`, the UDP listen port is assigned to `CHIP_PORT + 1`. If multiple processes are using the SDK, this can cause message delivery failures as both processes are bound to the same port.

 #### Proposed Solution

Either of:

* Fix this port to `0` to allow random selection.
* Allow propagating a value for this port from the Android Java layer.

andy31415, 2022-06-08:
@gmarcosb will look into this.

regarding deferral candidate, will leave it up to Marcos if this is important pre 1.0 or not given that this is platform-specific.

andy31415, 2022-06-08:
For now assuming not deferral since this is relating on Android platfor support for controllers.

g-coppock, 2022-06-08:
Yeah, notably since we expect multiple applications to include this SDK, I think the default behavior here will cause incompatibility. (e.g. if two apps A and B are installed with the default configuration, and both processes are running, then they will run into collissions).

PR #19596 (2022-06-23): Allow UDP listening port selection for Android
#### Problem
* Fixes #14542

#### Testing
Manually tested on android
closed 2022-06-23 by woody-apple