project-chip/connectedhomeip #14552: TE7.5 Mac Chip Tool testing - Windowcovering cluster config status returning 3 all the time even when mode attribute is being changed

Garik098, 2022-01-28:
While testing Windowcovering Cluster TC-WNCV-2.3 using Mac chip tool, Config status is returning 3 all the time even when mode attribute bits were being changed. 

I used the below command to set the different modes,

./out/debug/chip-tool windowcovering write mode 0001 1 1

changing 0001 to 0010, 0000, etc. The write is going through without any error, but the Config status is 3 all the time.

adk@adks-Mac-mini connectedhomeip % ./out/debug/chip-tool windowcovering read config-status 1 1
[1643408599924] [2175:115261] CHIP: [DL] _Init
[1643408599924] [2175:115261] CHIP: [IN] UDP::Init bind&listen port=5542
[1643408599925] [2175:115261] CHIP: [IN] UDP::Init bound to port=5542
[1643408599925] [2175:115261] CHIP: [IN] UDP::Init bind&listen port=5542
[1643408599925] [2175:115261] CHIP: [IN] UDP::Init bound to port=5542
[1643408599925] [2175:115261] CHIP: [IN] TransportMgr initialized
[1643408599925] [2175:115261] CHIP: [DIS] Init fabric pairing table with server storage
[1643408599925] [2175:115261] CHIP: [ZCL] Using ZAP configuration...
[1643408599925] [2175:115261] CHIP: [CTL] System State Initialized...
[1643408599927] [2175:115261] CHIP: [CTL] Generating NOC
[1643408599927] [2175:115261] CHIP: [CTL] Generating ICAC
[1643408599927] [2175:115261] CHIP: [DIS] Verifying the received credentials
[1643408599928] [2175:115261] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000001 and root pubkey
[1643408599928] [2175:115261] CHIP: [IN] 0x04, 0x30, 0xad, 0xd0, 0x70, 0x35, 0x89, 0x7a, 
[1643408599928] [2175:115261] CHIP: [IN] 0x44, 0x5a, 0x67, 0x52, 0x0a, 0xf3, 0xf3, 0xbd, 
[1643408599928] [2175:115261] CHIP: [IN] 0x3b, 0x4c, 0x98, 0x18, 0xb9, 0x0d, 0xf3, 0x4f, 
[1643408599928] [2175:115261] CHIP: [IN] 0x65, 0x03, 0xa4, 0xea, 0x46, 0xcf, 0x31, 0xc7, 
[1643408599928] [2175:115261] CHIP: [IN] 0x57, 0x47, 0x3a, 0x77, 0xa6, 0x45, 0x8f, 0x8d, 
[1643408599928] [2175:115261] CHIP: [IN] 0x39, 0xd8, 0xf2, 0xbc, 0xcd, 0x5d, 0x09, 0x07, 
[1643408599928] [2175:115261] CHIP: [IN] 0xab, 0x58, 0xa5, 0x18, 0x5e, 0xd1, 0x48, 0x10, 
[1643408599928] [2175:115261] CHIP: [IN] 0xe8, 0x14, 0xab, 0xd3, 0x8e, 0xe3, 0xca, 0x2e, 
[1643408599928] [2175:115261] CHIP: [IN] 0xee, 
[1643408599928] [2175:115261] CHIP: [IN] Generated compressed fabric ID
[1643408599928] [2175:115261] CHIP: [IN] 0xb5, 0x82, 0x93, 0xa5, 0x4b, 0x0a, 0x1b, 0xd9, 
[1643408599928] [2175:115261] CHIP: [DIS] Added new fabric at index: 1, Initialized: 1
[1643408599928] [2175:115261] CHIP: [DIS] Assigned compressed fabric ID: 0xB58293A54B0A1BD9, node ID: 0x000000000001B669
[1643408599928] [2175:115261] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0xB58293A54B0A1BD9
[1643408599928] [2175:115261] CHIP: [CTL] Generating NOC
[1643408599928] [2175:115261] CHIP: [CTL] Generating ICAC
[1643408599929] [2175:115261] CHIP: [DIS] Verifying the received credentials
[1643408599929] [2175:115261] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000002 and root pubkey
[1643408599929] [2175:115261] CHIP: [IN] 0x04, 0x30, 0xad, 0xd0, 0x70, 0x35, 0x89, 0x7a, 
[1643408599929] [2175:115261] CHIP: [IN] 0x44, 0x5a, 0x67, 0x52, 0x0a, 0xf3, 0xf3, 0xbd, 
[1643408599929] [2175:115261] CHIP: [IN] 0x3b, 0x4c, 0x98, 0x18, 0xb9, 0x0d, 0xf3, 0x4f, 
[1643408599929] [2175:115261] CHIP: [IN] 0x65, 0x03, 0xa4, 0xea, 0x46, 0xcf, 0x31, 0xc7, 
[1643408599929] [2175:115261] CHIP: [IN] 0x57, 0x47, 0x3a, 0x77, 0xa6, 0x45, 0x8f, 0x8d, 
[1643408599929] [2175:115261] CHIP: [IN] 0x39, 0xd8, 0xf2, 0xbc, 0xcd, 0x5d, 0x09, 0x07, 
[1643408599929] [2175:115261] CHIP: [IN] 0xab, 0x58, 0xa5, 0x18, 0x5e, 0xd1, 0x48, 0x10, 
[1643408599929] [2175:115261] CHIP: [IN] 0xe8, 0x14, 0xab, 0xd3, 0x8e, 0xe3, 0xca, 0x2e, 
[1643408599929] [2175:115261] CHIP: [IN] 0xee, 
[1643408599929] [2175:115261] CHIP: [IN] Generated compressed fabric ID
[1643408599929] [2175:115261] CHIP: [IN] 0x32, 0x34, 0xfd, 0x88, 0xd1, 0xfd, 0x2e, 0xad, 
[1643408599929] [2175:115261] CHIP: [DIS] Added new fabric at index: 2, Initialized: 1
[1643408599929] [2175:115261] CHIP: [DIS] Assigned compressed fabric ID: 0x3234FD88D1FD2EAD, node ID: 0x000000000001B669
[1643408599929] [2175:115261] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0x3234FD88D1FD2EAD
[1643408599930] [2175:115261] CHIP: [CTL] Generating NOC
[1643408599930] [2175:115261] CHIP: [CTL] Generating ICAC
[1643408599930] [2175:115261] CHIP: [DIS] Verifying the received credentials
[1643408599931] [2175:115261] CHIP: [IN] Generating compressed fabric ID using uncompressed fabric ID 0x0000000000000003 and root pubkey
[1643408599931] [2175:115261] CHIP: [IN] 0x04, 0x30, 0xad, 0xd0, 0x70, 0x35, 0x89, 0x7a, 
[1643408599931] [2175:115261] CHIP: [IN] 0x44, 0x5a, 0x67, 0x52, 0x0a, 0xf3, 0xf3, 0xbd, 
[1643408599931] [2175:115261] CHIP: [IN] 0x3b, 0x4c, 0x98, 0x18, 0xb9, 0x0d, 0xf3, 0x4f, 
[1643408599931] [2175:115261] CHIP: [IN] 0x65, 0x03, 0xa4, 0xea, 0x46, 0xcf, 0x31, 0xc7, 
[1643408599931] [2175:115261] CHIP: [IN] 0x57, 0x47, 0x3a, 0x77, 0xa6, 0x45, 0x8f, 0x8d, 
[1643408599931] [2175:115261] CHIP: [IN] 0x39, 0xd8, 0xf2, 0xbc, 0xcd, 0x5d, 0x09, 0x07, 
[1643408599931] [2175:115261] CHIP: [IN] 0xab, 0x58, 0xa5, 0x18, 0x5e, 0xd1, 0x48, 0x10, 
[1643408599931] [2175:115261] CHIP: [IN] 0xe8, 0x14, 0xab, 0xd3, 0x8e, 0xe3, 0xca, 0x2e, 
[1643408599931] [2175:115261] CHIP: [IN] 0xee, 
[1643408599931] [2175:115261] CHIP: [IN] Generated compressed fabric ID
[1643408599931] [2175:115261] CHIP: [IN] 0xf8, 0xc7, 0xb6, 0x6a, 0xf6, 0xc8, 0xb1, 0x3e, 
[1643408599931] [2175:115261] CHIP: [DIS] Added new fabric at index: 3, Initialized: 1
[... 99 lines cut ...]
[1643408599961] [2175:115264] CHIP: [DMG] 			
[1643408599961] [2175:115264] CHIP: [DMG] 		},
[1643408599961] [2175:115264] CHIP: [DMG] 		
[1643408599961] [2175:115264] CHIP: [DMG] 	],
[1643408599961] [2175:115264] CHIP: [DMG] 	
[1643408599961] [2175:115264] CHIP: [DMG] 	SuppressResponse = true, 
[1643408599961] [2175:115264] CHIP: [DMG] }
[1643408599961] [2175:115264] CHIP: [TOO] WindowCovering.ConfigStatus response: 3
[1643408599961] [2175:115264] CHIP: [EM] Sending Standalone Ack for MessageCounter:10543078 on exchange 36636i
[1643408599961] [2175:115264] CHIP: [IN] Prepared encrypted message 0x70000de04580 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 36636i with MessageCounter:12083918.
[1643408599961] [2175:115264] CHIP: [IN] Sending encrypted msg 0x70000de04580 with MessageCounter:12083918 to 0x0000000000000001 (1) at monotonic time: 13859722 msec
[1643408599961] [2175:115264] CHIP: [EM] Flushed pending ack for MessageCounter:10543078 on exchange 36636i
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the commissioner
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the controller
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the commissioner
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the controller
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the commissioner
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the controller
[1643408599961] [2175:115261] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1643408599961] [2175:115261] CHIP: [BLE] CancelConnection
[1643408599961] [2175:115261] CHIP: [DL] Inet Layer shutdown
[1643408599961] [2175:115261] CHIP: [DL] BLE shutdown
[1643408599961] [2175:115261] CHIP: [DL] System Layer shutdown

As per the test case steps, config status bits should change accordingly.

cjandhyala, 2022-05-17:
@Survensa could you pls verify this ?

Survensa, 2022-05-17:
Issue can be closed, verified with MAC and attched the log 

[WNCV.2.3.txt](https://github.com/project-chip/connectedhomeip/files/8708192/WNCV.2.3.txt)

cjandhyala, 2022-05-17:
@Survensa confirmed its not reproducible anymore, closing the ticket