project-chip/connectedhomeip #14555: mDNS results still delivered to a previously destroyed `DeviceController`

mrjerryjohns, 2022-01-28:
#### Problem

When doing multi-fabric tests in the Python REPL, shutting down a controller too quickly after commissioning results in:

```
[1643411841.267987][4346:4351] CHIP:CTL: Shutting down the commissioner
[1643411841.268007][4346:4351] CHIP:CTL: Shutting down the controller
[1643411841.268145][4346:4351] CHIP:CTL: Shutting down the commissioner
[1643411841.268426][4346:4351] CHIP:CTL: Shutting down the controller
[1643411841.268735][4346:4351] CHIP:DIS: Discovered node without a pending query
[1643411841.268773][4346:4351] CHIP:DIS: Node ID resolved for 0x0000000000000001
[1643411841.268791][4346:4351] CHIP:DIS:     Addr 0: [2001:db8:1::3]:5540
[1643411841.268804][4346:4351] CHIP:DIS:     Addr 1: [fe80::42:acff:fe14:3]:5540
[1643411841.268817][4346:4351] CHIP:DIS:     Addr 2: [172.20.0.3]:5540
[1643411841.268829][4346:4351] CHIP:DIS:     Addr 3: [2001:db8:1::3]:5540
[1643411841.268869][4346:4351] CHIP:DIS:     Addr 4: [fe80::42:acff:fe14:3]:5540

Thread 2 "python3" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4065700 (LWP 4351)]
0x00007fffe4087670 in ?? ()
#0  0x00007fffe4087670 in  ()
#1  0x00007ffff4aeeaee in chip::Dnssd::(anonymous namespace)::PacketDataReporter::OnComplete(mdns::Minimal::ActiveResolveAttempts&)
    (this=0x7ffff4064800, activeAttempts=...)
    at ../../src/lib/dnssd/Resolver_ImplMinimalMdns.cpp:334
#2  0x00007ffff4aeecf3 in chip::Dnssd::(anonymous namespace)::MinMdnsResolver::OnMdnsPacketData(mdns::Minimal::BytesRange const&, chip::Inet::IPPacketInfo const*)
    (this=0x7ffff4c2f820 <chip::Dnssd::(anonymous namespace)::gResolver>, data=..., info=0x7ffff4064c10)
    at ../../src/lib/dnssd/Resolver_ImplMinimalMdns.cpp:402
#3  0x00007ffff4bcbf01 in chip::Dnssd::GlobalMinimalMdnsServer::OnResponse(mdns::Minimal::BytesRange const&, chip::Inet::IPPacketInfo const*)
    (this=0x7ffff4c31180 <chip::Dnssd::GlobalMinimalMdnsServer::Instance()::_instance>, data=..., info=0x7ffff4064c10)
    at ../../src/lib/dnssd/MinimalMdnsServer.h:113
#4  0x00007ffff4b97936 in mdns::Minimal::ServerBase::OnUdpPacketReceived(chip::Inet::UDPEndPoint*, chip::System::PacketBufferHandle&&, chip::Inet::IPPacketInfo const*) (endPoint=0xe106c0, buffer=..., info=0x7ffff4064c10)
    at ../../src/lib/dnssd/minimal_mdns/Server.cpp:428
#5  0x00007ffff4bc5a88 in chip::Inet::UDPEndPointImplSockets::HandlePendingIO(chip::BitFlags<chip::System::SocketEventFlags, unsigned char>)
    (this=0xe106c0, events=...)
    at ../../src/inet/UDPEndPointImplSockets.cpp:688
#6  0x00007ffff4bc52f6 in chip::Inet::UDPEndPointImplSockets::HandlePendingIO(chip::BitFlags<chip::System::SocketEventFlags, unsigned char>, long)
    (events=..., data=14747328)
    at ../../src/inet/UDPEndPointImplSockets.cpp:569
#7  0x00007ffff4b0f255 in chip::System::LayerImplSelect::HandleEvents()
    (this=0x7ffff4c2fdc0 <chip::DeviceLayer::SystemLayerImpl()::gSystemLayerImpl>) at ../../src/system/SystemLayerImplSelect.cpp:402
#8  0x00007ffff4b2212c in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop()
    (this=0x7ffff4c30810 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>)
    at ../../src/include/platform/internal/GenericPlatformManagerImpl_POSIX.cpp:181
#9  0x00007ffff499cc86 in chip::DeviceLayer::PlatformManager::RunEventLoop()
    (this=0x7ffff4c30800 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../src/include/platform/PlatformManager.h:351
#10 0x00007ffff499ccc1 in (anonymous namespace)::PlatformMainLoop(void*) ()
    at ../../src/controller/python/chip/native/StackInit.cpp:36
#11 0x00007ffff7db5609 in start_thread ()
    at /lib/x86_64-linux-gnu/libpthread.so.0
#12 0x00007ffff7ef1293 in clone () at /lib/x86_64-linux-gnu/libc.so.6
```

mrjerryjohns, 2022-01-28:
FYI @vivien-apple @cecille

cecille, 2022-01-31:
Can you tell if shutdown is being called? OnMdnsPacketData does check that the delegate is not null before using it, and the shutdown on the resolver proxy sets that back to null.

bzbarsky-apple, 2022-02-18:
The problem is that shutting down the `ResolverProxy` doesn't unregister the delegate from the minimal-mdns singleton.  If you look in `Resolver_ImplMinimalMdns.cpp` it has things like:
```
CHIP_ERROR ResolverProxy::FindCommissionableNodes(DiscoveryFilter filter)
{
    VerifyOrReturnError(mDelegate != nullptr, CHIP_ERROR_INCORRECT_STATE);
    chip::Dnssd::Resolver::Instance().SetCommissioningDelegate(mDelegate);
    return chip::Dnssd::Resolver::Instance().FindCommissionableNodes(filter);
}
```
After this the `MinMdnsResolver` holds a pointer to the delegate.  And when `ResolverProxy::Shutdown` is called it does:
```
        mDelegate->Release();
        mDelegate = nullptr;
```
but the `MinMdnsResolver` is still holding a pointer to the now-dead object.

@andy31415

mrjerryjohns, 2022-06-23:
I had added `time.sleep(1)` in the Multi Fabric Python test to work around this issue. Let me remove it now and see if everything works fine...

mrjerryjohns, 2022-06-23:
Marking this as p1 and V1.0, since it's a fairly critical piece of machinery that needs to work right for multi fabric deployments.

bzbarsky-apple, 2022-06-23:
> I belive this is likely not happening anymore as AddressResolver is used 

It's not used for _FindCommissionableNodes_, which is what this issue was about.  That's using `mDNSResolver` in the `DeviceCommissioner`, which is a `ResolverProxy`.

andy31415, 2022-06-23:
I did not specifically fix for this issue, but I did update a lot of the discovery code. It is very likely gone (i fixed any lifetime bugs I could see when updating to address resolver) but I cannot guarantee it.

andy31415, 2022-06-23:
As far as I can tell the changes that likely fixed this are:

AbstractDnssdDiscoveryController::~AbstractDnssdDiscoveryController calls mDNSResolver.Shutdown()
Dnssd::ResolverProxy::Shutdown removes all delegates

closed 2022-06-24 by bzbarsky-apple