project-chip/connectedhomeip #14559: iOS tv-casting-app: Implement Content Launch command

sharadb-amazon, 2022-01-28:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 * actual behavior
 * steps to reproduce
 * system configuration
 * screenshots, images, logs, etc.
 >

 #### Proposed Solution
<suggested fix, suggested enhancement>

turon, 2022-02-01:
1.0 spec describes commands: `LaunchContent Request`, `LaunchURL Request`, and `Launch Response`.
These would presumably need to be implemented for v1 for the ContentLauncher cluster to be part of 1.0.
Removing tag `v1_triage_split_4`

woody-apple, 2022-02-09:
SDK supports this, but moving this tooling enhancement to post V1

sharadb-amazon, 2022-06-28:
Addressed in https://github.com/project-chip/connectedhomeip/pull/19779

closed 2022-06-28 by sharadb-amazon