project-chip/connectedhomeip #14566: If no attribute and event data encoded successfully, we should send status report instead of drop out

yunhanw-google, 2022-01-29:
Context is from

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/13400#discussion_r794897876_

Fix the logic here
        if (!hasEncodedAttributes && !hasEncodedEvents && hasMoreChunks)
        {
            ChipLogError(DataManagement,
                         "No data actually encoded but hasMoreChunks flag is set, abort report! (attribute too big?)");
            ExitNow(err = CHIP_ERROR_INCORRECT_STATE);
        }

bzbarsky-apple, 2022-01-29:
In particular, we should probably do a path-specific status for whatever path it was that did not fit in the packet.

bzbarsky-apple, 2022-02-15:
RESOURCE_EXHAUSTED seems like a good status to use here.

woody-apple, 2022-05-26:
Spec Review: @mrjerryjohns to review.

mrjerryjohns, 2022-06-22:
Chatted with @bzbarsky-apple , punting for post v1.0 since it's not really needed.