project-chip/connectedhomeip #14576: [[TC-MF-1.4] The coammand of ./chip-tool administratorcommissioning open-basic-commissioning-window always fails.

nanmu2018, 2022-01-29:
Problem：
[TC-MF-1.4] The coammand of ./chip-tool administratorcommissioning open-basic-commissioning-window always fails.

platform:
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba
[TC-MF-1.4.txt](https://github.com/project-chip/connectedhomeip/files/7963207/TC-MF-1.4.txt)
[otb-log.txt](https://github.com/project-chip/connectedhomeip/files/7963208/otbr-log.txt)
[light-log.txt](https://github.com/project-chip/connectedhomeip/files/7963211/light-log.txt)

cecille, 2022-02-25:
This command requires a timed interaction (see spec 11.19.8). You can specify that on the command line by adding the flag 
`--timedInteractionTimeoutMs` and the desired timeout.

Could you please try again with that flag set and report back?

bzbarsky-apple, 2022-05-05:
Works for me if I pass the right args...

closed 2022-05-05 by bzbarsky-apple