project-chip/connectedhomeip #14578: TE7.5 The third step of [TC-MF-1.22] pairing open-commissioning-window fails .

nanmu2018, 2022-01-29:
Problem：
The third step of [TC-MF-1.22] pairing open-commissioning-window fails .

platform:
Testing image on RasPi：v1.7.img
Connectedhomeip commitID : b0af6ba
OTBR commitID：72bb3d45684f837e67e7f1ec20a9a8e3ac4a4419
lighting-app run on the EFR32MG24 board base on commitID : b0af6ba
[TC-MF-1.22.txt](https://github.com/project-chip/connectedhomeip/files/7963225/TC-MF-1.22.txt)
[otbr-log.txt](htttps://github.com/project-chip/connectedhomeip/files/7963227/otbr-log.txt)
[light-log.txt](https://github.com/project-chip/connectedhomeip/files/7963230/light-log.txt)

cecille, 2022-02-25:
is this a dup of https://github.com/project-chip/connectedhomeip/issues/14576?

bzbarsky-apple, 2022-05-05:
No, this is different.  This is doing a timed interaction properly and is opening a commissioning window, per the log.  Then it's trying to open another commissioning window while the first one is open.  The Sigma handshake for that times out.

@Minan-Tuyasmart is this a problem on tip?

woody-apple, 2022-05-26:
Spec Review: @cjandhyala can you please have this re-tested?

cjandhyala, 2022-05-27:
@Minan-Tuyasmart Could you pls re-test this on latest ToT and report the result. If this is still an issue pls give full details on the platform , SHA, and the series of commands you executed

woody-apple, 2022-06-21:
Spec Issue Review: Given the age of this (TE 7.5), marking this as closed, given we have no open cert blockers on this.

closed 2022-06-21 by woody-apple