project-chip/connectedhomeip #14586: Ensure we reject operations on fabric scoped attributes when using PASE session

erjiaqing, 2022-01-31:
We absolutely want this, per current spec proposals.  Please make sure an issue is filed tracking this.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/13756#discussion_r794989381_

mlepage-google, 2022-01-31:
Well, sort of. This is my understanding.

When the PASE session is using accessing fabric index 0, it should not be able to use any fabric scoped attributes. When the PASE session gets elevated to the commissioned fabric as its accessing fabric (after AddNOC), it should be able to affect attributes scoped to that fabric (but not others).

bzbarsky-apple, 2022-01-31:
Yes, exactly.  This is not about PASE per se, but about whether we have an accessing fabric.  Unfortunately this issue omitted the context for my comment, which was code that said:
```
  // TODO: We may want to reject kUndefinedFabricIndex for writing fabric scoped data.
```

mlepage-google, 2022-02-10:
Just to clarify, access control is separate from other checks (but related of course). You might have permission to write an attribute, but it's not writable. You might have permission to write it, but it's scoped to another fabric. In those cases, you'll pass access control checks, but the operation (or at least that part of the operation) will be rejected.

Now, during commissioning, access control is granting permissions without an ACL, because you're coming in over a PASE session. This has two phases:

- Before AddNOC completes --> no accessing fabric
- After AddNOC completes --> accessing fabric is present

In the latter case, it's very similar to when you have an ACL over CASE, except it's over PASE and the access is still implicit (not from an actual ACL entry). You have a fabric, but it has to match. So e.g. if you read data, the data that does not match the fabric should be redacted.

In the former case, there is no accessing fabric. So operations will pass access control (with implicit admin privilege), but if you read data and the fabric does not match, it should be redacted. But without an accessing fabric, no such fabric-scoped data would ever match, so such data would always be redacted. And a fabric-scoped write would always fail.

The spec should say whether such operations silently fail, result in status, affect other operations that would not fail, etc. I'm thinking it would be very similar to when you do have an accessing fabric, but it's not the same one as the data.

It's possible that the code paths in place already handle this, but we need to audit to ensure this is covered, and of course we would like to have test coverage for this case, as it's a security concern.

bzbarsky-apple, 2022-02-11:
> The spec should say whether such operations silently fail, result in status, affect other operations that would not fail, etc

Yes, it should.  The spec around this part is nonsense; there are spec issues tracking it.

What should happen for this issue, probably, is that this function in AttributeValueDecoder:
```
    template <typename T, typename std::enable_if_t<DataModel::IsFabricScoped<T>::value, bool> = true>
    CHIP_ERROR Decode(T & aArg)
    {
        mTriedDecode = true;
        // TODO: We may want to reject kUndefinedFabricIndex for writing fabric scoped data. mAccessingFabricIndex will be
        // kUndefinedFabricIndex on PASE sessions.
        ReturnErrorOnFailure(DataModel::Decode(mReader, aArg));
        aArg.SetFabricIndex(AccessingFabricIndex());
        return CHIP_NO_ERROR;
    }
```
should in fact get that TODO comment addressed.

PR #15442 (2022-02-24): [IM] Ensure we reject operations on fabric scoped attributes when using PASE session before AddNOC
#### Problem

Fixes #14586 

#### Change overview

Addes a check to AttributeAccessInterface.h

#### Testing

The testing session comes with a fabric id,. so we add a override when accessing testing attribute.
closed 2022-02-24 by andy31415