project-chip/connectedhomeip #14590: [TC-SWDIAG-3.1] The value of CurrentHeapHighWatermark should match with the value of CurrentHeapUsed but failed to do so.

sowmyassp, 2022-01-31:
#### Problem
In [TC-SWDIAG-3.1] testcase fails because in step 2d the value of CurrentHeapHighWatermark should match with the value of CurrentHeapUsed in step 2c according to Spec.

But in master branch while running those commands the values are diiferent.
[TC_SWDIAG_3_1_manual.txt](https://github.com/project-chip/connectedhomeip/files/7971194/TC_SWDIAG_3_1_manual.txt)

tcarmelveilleux, 2022-01-31:
@sowmyassp Can you please provide the text of the steps you ran, so that we have the correct context?

sowmyassp, 2022-02-02:
@tcarmelveilleux ./chip-tool softwarediagnostics read current-heap-high-watermark 1 0
                             ./chip-tool  softwarediagnostics  read current-heap-used 1 0

yufengwangca, 2022-04-18:
It indicates we have memory leak

bzbarsky-apple, 2022-04-19:
Per Slack discussion, @yufengwangca found that `ReleaseObject` on the heap object pull unconditionally defers the deletion of the node even if we're not in the middle of iteration (i.e. even if `mIterationDepth` is 0).  For pools that never get iterated, or that don't ever do `ReleaseObject` from inside iteration, this ends up leaking the nodes...

yufengwangca, 2022-04-19:
https://github.com/project-chip/connectedhomeip/pull/17516 Fix part of memory leak, I can see the increment became less with this fix, but it shows there is still memory leak in our stack.

yufengwangca, 2022-04-28:
After tracking the memory leak, I found the leak is caused by the UnauthenticatedSession object allocated for each attribute transaction. 

It looks like the UnauthenticatedSession object is only allocated, but not released for later re-use. UnauthenticatedSessionTable is rotated by LRU.

yufengwangca, 2022-04-28:
@sowmyassp  please see my analysis above, current behavior reflects our current design.

sowmyassp, 2022-05-20:
Based on the comment given by jaehs6sam on issue([1238](https://github.com/CHIP-Specifications/chip-test-plans/issues/1238))I have reopened this iissue

woody-apple, 2022-05-23:
Cert Blocker Review: Assigning to @cjandhyala to update the test cases.

cjandhyala, 2022-05-24:
Closing the issue and followed up with a test plan ticket to update the TC

closed 2022-05-24 by cjandhyala

sowmyassp, 2022-05-24:
I have raised this ticket long back and later the test-plan is updated with test-case IDs so now this issue is for TC-DGSW-2.3, I will update the same in the heading, this ticket is valid @cjandhyala