project-chip/connectedhomeip #14595: Commissioning over BLE is failing on Master Branch

sowmyassp, 2022-01-31:
#### Problem
Commissioning over BLE is failing on Master Branch (CommitID -  8b503ad05993ccef1063c1b38d9a906a85955445 ).
[image]
[image]

Attached log : 
[pairing_failure_dut.txt](https://github.com/project-chip/connectedhomeip/files/7972508/pairing_failure_dut.txt)
[pairing_failure_cntrl.txt](https://github.com/project-chip/connectedhomeip/files/7972509/pairing_failure_cntrl.txt)

andy31415, 2022-01-31:
@sowmyassp - what platform are you running the commisioner on? Generally BLE differs greatly between mac/linux/android and in linux it may be useful to differentiate between RPi and ubuntu/other.

Also #14472 talks about core dumps.

andy31415, 2022-01-31:
Guessing the coredump is a sideffect of 'failed happy path' so the first thing we want to fix is probably the happy path bit.

cjandhyala, 2022-01-31:
@andy31415 I believe this happening on raspi platform.

cecille, 2022-01-31:
This is related to the network feature map changes.

```
[1643607818.011279][8535:8540] CHIP:CTL: Sending request for network cluster feature map
[1643607818.011667][8535:8540] CHIP:DMG: SendReadRequest ReadClient[0xffff78032000]: Sending Read Request
[1643607818.011922][8535:8540] CHIP:IN: Prepared secure message 0xffff81d327f8 to 0x0000000000000001 (1)  of type 0x2 and protocolId (0, 1) on exchange 3083i with MessageCounter:5750927.
[1643607818.012016][8535:8540] CHIP:IN: Sending encrypted msg 0xffff81d327f8 with MessageCounter:5750927 to 0x0000000000000001 (1) at monotonic time: 9829933 msec
[1643607818.012217][8535:8540] CHIP:DMG: MoveToState ReadClient[0xffff78032000]: Moving to [AwaitingIn]
[1643607818.012331][8535:8540] CHIP:DMG: ICR moving to [AwaitingDe]
[1643607818.103311][8535:8540] CHIP:DL: HandlePlatformSpecificBLEEvent 16387
[1643607818.105266][8535:8538] CHIP:DL: Indication received, conn = 0xffff7803ea30
[1643607818.105609][8535:8540] CHIP:DL: HandlePlatformSpecificBLEEvent 16389
[1643607818.105931][8535:8540] CHIP:EM: Received message of type 0x5 with protocolId (0, 1) and MessageCounter:5197166 on exchange 3083i
[1643607818.106010][8535:8540] CHIP:EM: Found matching exchange: 3083i, Delegate: 0xffff78032000
[1643607818.106139][8535:8540] CHIP:DMG: ReportDataMessage =
[1643607818.106204][8535:8540] CHIP:DMG: {
[1643607818.106258][8535:8540] CHIP:DMG: 	AttributeReportIBs =
[1643607818.106334][8535:8540] CHIP:DMG: 	[
[1643607818.106395][8535:8540] CHIP:DMG: 		AttributeReportIB =
[1643607818.106481][8535:8540] CHIP:DMG: 		{
[1643607818.106545][8535:8540] CHIP:DMG: 			AttributeDataIB =
[1643607818.106631][8535:8540] CHIP:DMG: 			{
[1643607818.106765][8535:8540] CHIP:DMG: 				DataVersion = 0x0,
[1643607818.106843][8535:8540] CHIP:DMG: 				AttributePathIB =
[1643607818.106924][8535:8540] CHIP:DMG: 				{
[1643607818.107016][8535:8540] CHIP:DMG: 					Endpoint = 0x0,
[1643607818.107104][8535:8540] CHIP:DMG: 					Cluster = 0x31,
[1643607818.107200][8535:8540] CHIP:DMG: 					Attribute = 0x0000_FFFC,
[1643607818.107284][8535:8540] CHIP:DMG: 				}
[1643607818.107378][8535:8540] CHIP:DMG: 					
[1643607818.107456][8535:8540] CHIP:DMG: 					Data = 2, 
[1643607818.107545][8535:8540] CHIP:DMG: 			},
[1643607818.107630][8535:8540] CHIP:DMG: 			
[1643607818.107697][8535:8540] CHIP:DMG: 		},
[1643607818.107772][8535:8540] CHIP:DMG: 		
[1643607818.107830][8535:8540] CHIP:DMG: 	],
[1643607818.107905][8535:8540] CHIP:DMG: 	
[1643607818.107965][8535:8540] CHIP:DMG: 	SuppressResponse = true, 
[1643607818.108028][8535:8540] CHIP:DMG: 	InteractionModelRevision = 1
[1643607818.108085][8535:8540] CHIP:DMG: 
```
^ This is the feature map query for the network cluster. The DUT is indicating that it is thread device. If the DUT is wifi, it needs to have the zap file updated to reflect that.

andy31415, 2022-01-31:
Where is this default value coming from?
I imagine it is moved from zap into each app, which makes it hard from shared zap files (all clusters works on both thread and wifi devices) to override.

I would like to surface this info in .matter IDL files to make it easy to review, but I am totally unclear how the default value makes its way into responses :(

andy31415, 2022-01-31:
I am not even clear it is zap: https://github.com/project-chip/connectedhomeip/blob/master/examples/all-clusters-app/all-clusters-common/all-clusters-app.zap#L2314 says `"defaultValue": 1` where as thread constant is 0x02

jmartinez-silabs, 2022-01-31:
I was also under the impression that this is the defaul value. and the feature map is set to 1 (wifi) for in all zap configs I saw. fwiw, I get the same error with lighting app on thread platform.

cecille, 2022-02-01:
What are the platform and app being used here?

sankarselvam, 2022-02-01:
Its happening on both Thread and Raspi Linux platform.
Apps used, Lighting-app in Thread  & All clusters app in Raspi platform

cecille, 2022-02-02:
Please see slack discussion https://csamembers.slack.com/archives/G014G30SVV0/p1643655728185279 for more context. 

Network commissioning issue is tracked here: 
https://github.com/project-chip/connectedhomeip/issues/13170

andy31415, 2022-02-02:
I believe it may be fixed - at least for linux and esp32, thread will need some tests and maybe fixes and I need to setup a test environment for that.

sowmyassp, 2022-02-09:
Commissioning over BLE is working now with recent commitID - cab1ff9cf129ee7d2b79c225a4682fca2f0bc1e8

closed 2022-02-09 by sowmyassp