project-chip/connectedhomeip #14596: Incorrect processing in Engine::ScheduleRun()

lnikulin, 2022-01-31:
#### Problem
Received timeout of operation on get/set attribute value into any clusters (any attributes) from client application (chip-tool), 

 #### Proposed Solution
setup value ```mRunScheduled = true``` in ```Engine::ScheduleRun()``` before
```
systemLayer->ScheduleWork(Run, this);
```
ScheduleWork can be called safely on any thread without locking the stack.  When called from a thread that is not doing the stack work item processing, the callback function may be called (on the work item processing thread) before ScheduleWork returns.
And as result in case callback function has been done before ScheduleWork returns, our flag mRunScheduled will always true on the next calls.
Simple fix:
[simple-fix.zip](https://github.com/project-chip/connectedhomeip/files/7972450/simple-fix.zip)

mrjerryjohns, 2022-02-02:
This all assumes that `ScheduleRun` is safe to call from a threading context other than Chip's, which is not explicitly stipulated nor correct. Unless explicitly stated, all SDK APIs **must** be called from within the Chip thread context itself.

This specific call gives the impression that it's ok to call from a different threading context but in fact, it actually isn't. It accesses a number of SDK state that can change on it during a shutdown (like the `InteractionModelEngine` singleton, the underlying `SessionManager` and SystemLayer objects). None of these are accessed safely with a mutex relative to the CHIP thread, so it's not safe to call this API from outside the CHIP threading context.

bzbarsky-apple, 2022-02-02:
Specifically, what would be most useful here is a stack to the `ScheduleRun` call involved.

yufengwangca, 2022-02-02:
@lnikulin Do we have the stack dump when the issue occurs? wondering which thread other than CHIP main thread is calling Engine::ScheduleRun()

lnikulin, 2022-02-03:
> Specifically, what would be most useful here is a stack to the `ScheduleRun` call involved.

lnikulin, 2022-02-03:
Call Stack look like following:
on Linux platform
```
[1643380890.231073][2200:2200] CHIP:DMG: <ThreadID: 139932194578816> SchedWork() begin
[1643380890.231078][2200:2200] CHIP:DMG: <ThreadID: 139932194578816> Signal begin
[1643380890.231083][2200:2200] CHIP:DMG: <ThreadID: 139932194578816> Signal end
[1643380890.231086][2200:2200] CHIP:DMG: <ThreadID: 139932194578816> SchedWork() end
[1643380890.231088][2200:2203] CHIP:DMG: <ThreadID: 139932106483456> Engine::Run() begin
[1643380890.231106][2200:2203] CHIP:DMG: <ThreadID: 139932106483456> Run() begin
[1643380890.231110][2200:2203] CHIP:DMG: <ThreadID: 139932106483456> Run() end
[1643380890.231115][2200:2203] CHIP:DMG: <ThreadID: 139932106483456> Engine::Run() end
```
on MIPS platform
```
[1643382659.193236][1927:1927] CHIP:DMG: <ThreadID: 2012581508> call SchedWork
[1643382659.193314][1927:1927] CHIP:DMG: <ThreadID: 2012581508> SchedWork() begin
[1643382659.193372][1927:1927] CHIP:DMG: <ThreadID: 2012581508> Signal begin
[1643382659.193493][1927:1930] CHIP:DMG: <ThreadID: 2006711656> Engine::Run() begin
[1643382659.193547][1927:1930] CHIP:DMG: <ThreadID: 2006711656> Run() begin
[1643382659.193592][1927:1930] CHIP:DMG: <ThreadID: 2006711656> Run() end
[1643382659.193633][1927:1930] CHIP:DMG: <ThreadID: 2006711656> Engine::Run() end
[1643382659.193744][1927:1927] CHIP:DMG: <ThreadID: 2012581508> Signal end
[1643382659.193795][1927:1927] CHIP:DMG: <ThreadID: 2012581508> SchedWork() end
```

bzbarsky-apple, 2022-02-04:
Where are the "SchedWork() begin" and "Signal begin" prints coming from?  That is, what code locations?

lnikulin, 2022-02-04:
File ```src/system/SystemLayerImplSelect.cpp```
```
CHIP_ERROR LayerImplSelect::ScheduleWork(TimerCompleteCallback onComplete, void * appState)
{
    ChipLogDetail(DataManagement, "<ThreadID: %lu><%s> SchedWork() begin", pthread_self());
    VerifyOrReturnError(mLayerState.IsInitialized(), CHIP_ERROR_INCORRECT_STATE);

    CancelTimer(onComplete, appState);

    TimerList::Node * timer = mTimerPool.Create(*this, SystemClock().GetMonotonicTimestamp(), onComplete, appState);
    VerifyOrReturnError(timer != nullptr, CHIP_ERROR_NO_MEMORY);

#if CHIP_SYSTEM_CONFIG_USE_DISPATCH
    dispatch_queue_t dispatchQueue = GetDispatchQueue();
    if (dispatchQueue)
    {
        (void) mTimerList.Add(timer);
        dispatch_async(dispatchQueue, ^{
            this->HandleTimerComplete(timer);
        });
        return CHIP_NO_ERROR;
    }
#endif // CHIP_SYSTEM_CONFIG_USE_DISPATCH

    if (mTimerList.Add(timer) == timer)
    {
        // The new timer is the earliest, so the time until the next event has probably changed.
        ChipLogDetail(DataManagement, "<ThreadID: %lu> Signal begin", pthread_self());
        Signal();
        ChipLogDetail(DataManagement, "<ThreadID: %lu> Signal end", pthread_self());
    }
    ChipLogDetail(DataManagement, "<ThreadID: %lu> SchedWork() end", pthread_self());
    return CHIP_NO_ERROR;
}
```

bzbarsky-apple, 2022-02-04:
OK, but that's not what we were asking.  We are asking: what code is calling `ScheduleRun`, not what code does `ScheduleRun` call.

Someone is calling `ScheduleRun` without synchronization on the wrong thread.  The question is who that is.

lnikulin, 2022-02-07:
Setup attribute emberAfWriteAttribute() from application thread (for example setup UniqueID into Basic Cluster EP0)

BSB002:
Main application thread = 2012425860
Chip Event loop thread  = 2006556008
```
[1644241241.522429][1927:1927] CHIP:DMG: <ThreadID: 2012425860> SchedWork() begin
[1644241241.522488][1927:1927] CHIP:DMG: <ThreadID: 2012425860> Signal begin
[1644241241.522604][1927:1930] CHIP:DMG: <ThreadID: 2006556008> Engine::Run() START
[1644241241.522654][1927:1930] CHIP:DMG: <ThreadID: 2006556008> Run() begin, set mRunScheduled = 0
[1644241241.522728][1927:1930] CHIP:DMG: <ThreadID: 2006556008> Run() end
[1644241241.522768][1927:1930] CHIP:DMG: <ThreadID: 2006556008> Engine::Run() END
[1644241241.522880][1927:1927] CHIP:DMG: <ThreadID: 2012425860> Signal end
[1644241241.522928][1927:1927] CHIP:DMG: <ThreadID: 2012425860> SchedWork() end
[1644241241.522970][1927:1927] CHIP:DMG: <ThreadID: 2012425860> ScheduleRun() setup mRunSched = 1
```
Linux:
Main application thread = 139729208574336
Chip Event loop thread  = 139729191761664
```
[1644239559.611446][404:404] CHIP:DMG: <ThreadID: 139729208574336> ScheduleRun call SchedWork
[1644239559.611452][404:404] CHIP:DMG: <ThreadID: 139729208574336> SchedWork() begin
[1644239559.611458][404:404] CHIP:DMG: <ThreadID: 139729208574336> Signal begin
[1644239559.611465][404:404] CHIP:DMG: <ThreadID: 139729208574336> Signal end
[1644239559.611469][404:404] CHIP:DMG: <ThreadID: 139729208574336> SchedWork() end
[1644239559.611474][404:404] CHIP:DMG: <ThreadID: 139729208574336> ScheduleRun setup mRunSched = 1
[1644239559.611474][404:407] CHIP:DMG: <ThreadID: 139729191761664> Engine::Run() START
[1644239559.611491][404:407] CHIP:DMG: <ThreadID: 139729191761664> Run() begin, set mRunScheduled = 0
[1644239559.611497][404:407] CHIP:DMG: <ThreadID: 139729191761664> Run() end
[1644239559.611502][404:407] CHIP:DMG: <ThreadID: 139729191761664> Engine::Run() END
```

bzbarsky-apple, 2022-02-07:
> Write attribute emberAfWriteAttribute() from application thread

Ok, this is the thing that's not allowed.  This needs to happen off a task queued on the Matter event loop.  Otherwise you can race not only with Engine::ScheduleRun but also with writes to attributes happening as a result of Matter message processing, so can end up with a corrupted attribute store.

bzbarsky-apple, 2022-02-08:
> We could wrap emberAfWriteAttribute() call with ScheduleWork

Yes, that is what you should do.

> However, emberAfWriteAttribute() call itself ends up in ScheduleRun which calls ScheduleWork at the end

This is after it has already done a bunch of work, like writing to the attribute store, and the remaining (async) work is sending subscription updates as needed.

> So, it looks like API designed to be thread agnostic

It's not thread agnostic at all.

lnikulin, 2022-02-09:
In case, we use wrapper for ```emberWriteAttribute``` and call it through the ```chip::DeviceLayer::PlatformMgr().ScheduleWork()```, our application works without any errors. I'll close this issue.