project-chip/connectedhomeip #14604: Enforce that no PASE sessions that aren't commissioning are possible

mlepage-google, 2022-01-31:
For v1, we do not support operational PASE, so PASE is only permissible during commissioning.

In access control, for v1 we grant any PASE session the administer privilege, since it's appropriate given commissioning can be taken for granted.

**Therefore, we need to enforce in the SDK that in v1 it is not possible to create a non-commissioning PASE session.** This may already be the case but we need to double check.

cecille, 2022-05-16:
PASE session establishment is only allowed once the commissioning window manager calls WaitForPairing. We already expire PASE pairings on commissioning complete, but we should also expire them when the commissioning window closes so that other nodes can't establish a long-running PASE connection and send commands after the commissioning window is closed.

woody-apple, 2022-11-02:
Issue Scrub: We believe in the current implementation of the SDK, PASE can't be easily, or at all used. Closing.

closed 2022-11-02 by woody-apple