project-chip/connectedhomeip #14612: [TC_IDM_4.1] & [TC_IDM_4.2] Need support for activating various subscription attribute types for IDM testing for TE7.5

liamgonyea, 2022-02-01:
#### Problem
To run test cases `[TC_IDM_4.1]` and `[TC_IDM_4.2]`, there needs to be a functionality for a user to activate subscriptions and send subscription request messages for individual subscription data types (bool, string, unsigned integer, signed integer, floating point)  from a chip-tool controller to an accessory on an individual basis.

#### Test Case
3.5.1. [TC-IDM-4.1] SubscriptionRequestMessage from DUT test cases
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#351-tc-idm-41-subscriptionrequestmessage-from-dut-test-cases

3.6.1. [TC-IDM-4.2] ReportDataMessage from DUT test cases
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#361-tc-idm-42-reportdatamessage-from-dut-test-cases

#### Testing Setup
TE7.5 commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`

vivien-apple, 2022-02-02:
Can you tell me exactly what you need ? `chip-tool` can subscribe to any attribute type already, but obviously you will need to call `subscribe` on the correct attributes based on the types from the spec.

woody-apple, 2022-03-03:
We believe this can be done by picking an attribute of each type on a cluster to confirm this, moving post v1.

mrjerryjohns, 2022-05-26:
This is correct, the test plan just needs to pick the right attributes from the various utility clusters that are known to be present on all devices to ensure it hits the gamut of types required by the test plan.