project-chip/connectedhomeip #14624: Events being sent on all the endpoints

chiragatal, 2022-02-01:
#### Problem
 * expected behavior: Events triggered on the device should be sent only to a particular endpoint_id.
 * actual behavior: Events triggered are sent to all the endpoints that have the cluster included.
 * steps to reproduce: Create a device with 2 endpoints and both having the switch cluster. Trigger the SwitchLatched event on the device. The event gets reported on both the endpoints instead of just one.
 * reference: Check `OnSwitchLatched` in `switch-server.cpp`.

 #### Proposed Solution
 * The events should accept the endpoint_id also as an argument, and send the event just on that endpoint.

dhrishi, 2022-02-01:
@yufengwangca Can you please check?

yufengwangca, 2022-02-02:
@chiragatal The event is triggered on the device, it does not have the endpoint_id as an argument, so it gets logged to all endpoints on which the relevant cluster is enabled

bzbarsky-apple, 2022-02-03:
The point is, the switch cluster API here is broken, no?  It should either know which endpoint is being addressed to start with (have per-endpoint delegates) or allow passing in the endpoint id from the caller.

Looks like the implementation in PR #13036 just does not make sense for a device with more than one switch on it....

dhrishi, 2022-02-03:
The same is to be checked/implemented for all other utility/app clusters too that support events?

bzbarsky-apple, 2022-02-03:
Probably worth it, yes...

mrjerryjohns, 2022-02-10:
Taking off the IM label since this isn't an IM limitation but rather an issue with the app clusters.

PR #18742 (2022-05-25): Refactor the switch delegate APIs from platform layer to application domain
#### Problem
What is being fixed?  Examples:
* expected behavior: Events triggered on the device should be sent only to a particular endpoint_id.
actual behavior: Events triggered are sent to all the endpoints that have the cluster included.
* Fixes #14624

#### Change overview
Refactor the switch delegate APIs from platform layer to application domain and event is only triggered on a particular endpoint_id.

TODO: Implement switch event trigger on one of existing example app

#### Testing
How was this tested? (at leas [... cut]
closed 2022-05-25 by bzbarsky-apple