project-chip/connectedhomeip #14628: Relative Humidity - In TC-RH- 1.1 the cluster revision value is  not matched with the expected outcome

somu1710, 2022-02-01:
In  Relativehumidity testcase 1.1 , they mentioned the cluster revision attribute value as 3  in the expected outcome but while executing the step it shows cluster revision value as 2.

 * expected behavior - cluster revision attribute - 3
 * actual behavior - cluster revision - 2
 
Actual command -  ./chip-tool relativehumiditymeasurement read cluster-revision 3 1
Actual logs are attached below 
[Relative humidity  1.1 DUT log.txt](https://github.com/project-chip/connectedhomeip/files/7978753/Relative.humidity.1.1.DUT.log.txt)
[Relative humidity 1.1 ctrl.txt](https://github.com/project-chip/connectedhomeip/files/7978845/Relative.humidity.1.1.ctrl.txt)

somu1710, 2022-04-26:
For Relative humidity cluster still the cluster revision value is not matched with the spec.

vivien-apple, 2022-05-18:
Duplicate of #17532

closed 2022-05-18 by vivien-apple