project-chip/connectedhomeip #14696: Command fails for ECM (using pairing open-commissioning-window) using CHIP TOOL for TE7.5

PSONALl, 2022-02-02:
#### Problem

Expected behavior:
pairing open-commissioning-window command should pass
Actual behavior:
pairing open-commissioning-window command fails with timeout issue on chip tool
Steps to reproduce:
Run steps as per test case: 3.1.3. [TC-MF-1.3] Node Behavior using ECM [DUT - Commissionee]
System configuration:
Chip tool of te7.5 rpi image on RPi 4B
DUT: esp32-devkit-c (connected to MacOS Catalina: 10.15.7)
App: all_clusters_app
CHIP repo: (master - b0af6ba42b6ce494748f18ecf3e054a400b8f45b)
\<logs attached\
[multi-admin-ecm-esp32.log](https://github.com/project-chip/connectedhomeip/files/7985941/multi-admin-ecm-esp32.log)
[multi-admin-ecm-chip-tool.log](https://github.com/project-chip/connectedhomeip/files/7985939/multi-admin-ecm-chip-tool.log)
>

jamesluo11, 2022-02-21:
Hi  guys, i just found the walkaround , i increate the waitDuration to 15s from 5s in examples/chip-tool/commands/common/CHIPCommand.cpp. the command return successfully.

PSONALl, 2022-02-23:
> Hi guys, i just found the walkaround , i increate the waitDuration to 15s from 5s in examples/chip-tool/commands/common/CHIPCommand.cpp. the command return successfully.

thanks @jamesluo11, I tried it at my end for esp32, It worked for me too!

jamesluo11, 2022-02-23:
@PSONALl No problem, anyway it's just open-commissioning-window case, other cases also have similar issue. I think we should work out solution to resove this kind of issue thoroughly.
@tcarmelveilleux FYI.

closed 2022-06-13 by bzbarsky-apple