project-chip/connectedhomeip #14709: [TC_DD_3.10] Need support for Custom Flow commissioning on sample accessory

liamgonyea, 2022-02-02:
#### Problem
When running [TC_DD_3.10], Steps 1-4, there needs to be support for a sample accessory, such as `chip-all-clusters-app`, to allow for Custom Flow commissioning of a device to a controller, such as `chip-tool` so that the end-to-end commissioning flow can properly be tested when using a controller DUT.

#### Test Case
3.3.10. [TC-DD-3.10] Commissioning Flow - Custom Flow [DUT Commissioner]
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicediscovery.adoc#3310-tc-dd-310-commissioning-flow---custom-flow-dut-commissioner

#### Test Setup
TE7.5 commit: `b0af6ba42b6ce494748f18ecf3e054a400b8f45b`

liamgonyea, 2022-03-17:
Verified support is still needed for User Intent and Custom Commissioning Flows while testing TE8 with commit `448578dd09004679d10a7fa883f2e680de05079d`using `chip-tool-darwin` and `chip-all-clusters-app`

Additional Test Cases that apply:
[TC-DD-3.11] Commissioning Flow - QR Code [DUT - Commissioner]

franck-apple, 2022-03-24:
Specs only specifies that the Custom Flow bit of the QR code is recognized as Custom commissioning flow. 
What's the ask here then? The development of a parser?

sowmyassp, 2022-03-31:
@franck-apple Currently we need support for custom flow commissioning on sample accessory but we have support for TH accessory.

vivien-apple, 2022-04-13:
@liamgonyea @sowmyassp  Sorry, the request here is unclear to me. My understanding of those test steps are:
  * DUT receives Onboarding Payload from the TH with Custom Flow bits set to value 2 -> Does not mean TH "send" an onboarding payload with the "Custom Flow" bits set to 2, but that there is an onboarding payload "somewhere" that is configured to have the custom flow bits set.
 Once the DUT reads such a payload it needs to do specific things such as checking the DCL to know how to put the commission the accessory. For example those instructions could be:
  * Download something at https://....
  * Go to the settings menu onto the accessory and select "Enable commissioning"
In reality, and this is only my understanding, those steps does not really requires any interaction with the TH itself - we just need to check that the DUT is providing instructions and the necessary tools to commission properly.

Maybe I misunderstood the test plan or the request, and in this case please correct me so I can help !

sowmyassp, 2022-05-13:
Hi @liamgonyea Test-case says we have to set custom-flow value, we can set the custom-flow value by passing command  "--custom-flow 0" , If this correct way to set can you please the issue.

woody-apple, 2022-05-26:
Spec Review: @vivien-apple can you please confirm if this is a spec issue?

woody-apple, 2022-05-26:
(And close if necessary)

vivien-apple, 2022-07-14:
> Hi @liamgonyea Test-case says we have to set custom-flow value, we can set the custom-flow value by passing command "--custom-flow 0" , If this correct way to set can you please the issue.

The only effect of the `--custom-flow 0` parameter onto the sample accessory afaik is to generate an appropriate QRCode/Manual Code and dump it on the console.

> Spec Review: @vivien-apple can you please confirm if this is a spec issue?

This is not a spec issue to me.

It is still unclear to me what kind of supports is expected. I would first like to make sure my understanding of the custom commissioning flow as expressed in https://github.com/project-chip/connectedhomeip/issues/14709#issuecomment-1097776896 is correct. And if not, what kind of support is expected.

Maybe a concrete example of an artificial custom commissioning flow can be provided ?

woody-apple, 2022-07-22:
Spec Issue Review: @vivien-apple should we close this then?

closed 2022-07-27 by woody-apple