project-chip/connectedhomeip #14725: Sort out network commissioning for ethernet over BLE

cecille, 2022-02-02:
#### Problem
We don't currently support commissioning ethernet devices that are not already on the network

 #### Proposed Solution
I guess we should allow that?

cecille, 2022-02-02:
see https://github.com/project-chip/connectedhomeip/pull/14567#discussion_r797781392

bzbarsky-apple, 2022-02-03:
I guess we would need to do ConnectNetwork to turn on the Ethernet port?  It's not 100% clear to me what the actual commissioning steps involved here are...

@jmeg-sfy what would you like to see here?

cecille, 2022-02-03:
I think the use case here is a device that either needs permission to enable the wired connection or permission to use it for matter operations. I'm not sure how common that use case would be, but I suppose it's technically possible.

I think the steps here would be to have the device read the network cluster "Networks" attribute to get the name of the ethernet network and then call enable on it.

From a commissioning side, this doesn't seem TOO hard, but the real challenge is how to test it.

jmeg-sfy, 2022-02-03:
> I guess we would need to do ConnectNetwork to turn on the Ethernet port? It's not 100% clear to me what the actual commissioning steps involved here are...
> 
> @jmeg-sfy what would you like to see here?

I don't request changes , just wondering how this would work in case an ETH port is off to save power

cecille, 2022-03-29:
Suggest we defer this to post 1.0 given that we don't have any devices that do this and don't have a solid use case.