project-chip/connectedhomeip #14757: 2nd-fabric commissioning sometimes times out establishing CASE

mrjerryjohns, 2022-02-03:
#### Problem

Cirque recently added a multi-fabric test that commissions (in-process) a target device into two independent fabrics.

Occasionally, it is failing to successfully commission the 2nd fabric. This does not happen all the time.

mrjerryjohns, 2022-02-03:
So managed to repro this locally with enhanced logging (logs can be found [here](https://github.com/project-chip/connectedhomeip/files/7997678/logs.1.txt)) as well as increasing the time it takes to process Sigma1 on the device to excaercerbate the failure, and here is the failure sequence:

Pre-Error State:
1. Have setup a `DeviceController` instance that is tracking Fabric 1 (Compressed Fabric ID: CF6514318CEA506039), with a NodeId of 1 for the controller.
2. That controller has successfully commissioned a target with NodeId 2.
3. Target is on the local network (commissioned over IP).
4. Successfully established a CASE session to that target.
5. Have setup a `DeviceController` instance that is tracking Fabric 2 (Compressed Fabric ID: CF89B0FF04E103554C) with a NodeId of 1 for the controller.

1. F2's Controller sends the NOC to the target:

```
2022-02-03 19:38:18 johnsj chip.CTL[7101] INFO Performing next commissioning step 'SendNOC' with completion status = '../../src/controller/CHIPDeviceController.cpp:1359: Success'
2022-02-03 19:38:18 johnsj chip.CTL[7101] INFO Sending operational certificate chain to the device
```

2. Before we get a response back to the SendNOC command, mDNS receives an unsolicited service announcement from the server (presumably because it just got commissioned and proceeded to broadcast immediately before it sent out the SendNOC response):

```
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO Node ID resolved for 0x0000000000000002
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 0: [fe80::a6ae:11ff:fe18:698a]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 1: [192.168.0.141]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 2: [fe80::a6ae:11ff:fe18:698a]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 3: [192.168.0.141]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 4: [fe80::a6ae:11ff:fe18:698a]:5540
2022-02-03 19:38:18 johnsj chip.CTL[7101] INFO 1: OperationalDiscoveryComplete for device ID 0x0000000000000002
```

The `ResolverProxy` instance within the `DeviceController` object (see `AbstractDnssdDiscoveryController::mDNSResolver`) has been previously registered as **the** delegate to route resolution responses back from the global resolver singleton when it had last been used to resolve a NodeId in the first commissioning attempt. This registration happens in `ResolverProxy::ResolveNodeId`, where it calls `chip::Dnssd::Resolver::Instance().SetResolverDelegate(mDelegate)`. 

Consequently, F1's controller get's notified first of the NodeID resolution. 

3. The logic in the controller attempts to always connect over CASE to any Nodes it gets notified of. In this case, F1's controller attempts to connect to the announced NodeId, which is actually on Fabric 2!

Since the `CASESessionManager` matches existing sessions against PeerId, it won't find a match for Fabric 2's NodeIds since it's only tracking Fabric 1's NodeIds in instance within F1's controller:
```
2022-02-03 19:38:18 johnsj chip.CSM[7101] DEBUG FindOrEstablishSession: PeerId = CF89B0FF04E103554C:0000000000000002}, NodeIdResolved = 1
2022-02-03 19:38:18 johnsj chip.CSM[7101] DEBUG FindOrEstablishSession: No existing session found
```

That commences CASE session establishment, which actually progresses despite emanating from F1's controller since the FabricInfo table is shared, and the appropriate creds can be picked up interestingly.

4. F2's controller's state machine eventually gets around the `FindOperational` state, and commences NodeId resolution.

The response comes back immediately (presumably from cache?) and like F1's controller, doesn't find a matching session to that Node and attempts to initiate CASE:

```
2022-02-03 19:38:18 johnsj chip.CTL[7101] INFO Going from commissioning step 'SendNOC' with lastErr = '../../src/controller/CHIPDeviceController.cpp:1383: Success' --> 'FindOperational'
2022-02-03 19:38:18 johnsj chip.CTL[7101] INFO Performing next commissioning step 'FindOperational' with completion status = '../../src/controller/CHIPDeviceController.cpp:1383: Success'
2022-02-03 19:38:18 johnsj chip.DIS[7101] ERROR Attempt to mDNS broadcast failed:  ../../src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000065: Network is unreachable
2022-02-03 19:38:18 johnsj chip.DMG[7101] DEBUG ICR moving to [AwaitingDe]
2022-02-03 19:38:18 johnsj chip.EM[7101] DEBUG Sending Standalone Ack for MessageCounter:1562338 on exchange 55025i
2022-02-03 19:38:18 johnsj chip.IN[7101] INFO Prepared secure message 0x7f86667fad68 to 0x0000000000000002 (2)  of type 0x10 and protocolId (0, 0) on exchange 55025i with MessageCounter:12628874.
2022-02-03 19:38:18 johnsj chip.IN[7101] INFO Sending encrypted msg 0x7f86667fad68 with MessageCounter:12628874 to 0x0000000000000002 (2) at monotonic time: 604373 msec
2022-02-03 19:38:18 johnsj chip.EM[7101] DEBUG Flushed pending ack for MessageCounter:1562338 on exchange 55025i
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO Node ID resolved for 0x0000000000000002
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 0: [fe80::a6ae:11ff:fe18:698a]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 1: [192.168.0.141]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 2: [fe80::a6ae:11ff:fe18:698a]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 3: [192.168.0.141]:5540
2022-02-03 19:38:18 johnsj chip.DIS[7101] INFO     Addr 4: [fe80::a6ae:11ff:fe18:698a]:5540
[... cut ...]
5. Because of the delay I added in processing Sigma1 on the device, that ensures we actually create a scenario where we have two CASE sessions being attempted at the same time.

Because the server restricts processing multiple CASE session attempts at the same time, the 2nd attempt from F2's controller actually gets rejected.

This results in that eventually timing out.

```
2022-02-03 19:38:24 johnsj chip.EM[7101] ERROR Failed to Send CHIP MessageCounter:1275174245 on exchange 55027i sendCount: 3 max retries: 3
```

That eventually causes the commissioning failure on F2's controller.

mrjerryjohns, 2022-02-03:
Interestingly, I cannot repro this failure on Mac. I think it has something to do with with the way DNS-SD announcements are made, and how many of them are made (perhaps as a function of the # of interfaces the server is being bound to?)

cecille, 2022-02-25:
@mrjerryjohns  - does https://github.com/project-chip/connectedhomeip/pull/15445 fix this issue?

mrjerryjohns, 2022-03-15:
I think this is fixed, since @drempelg ran into this issue as well recently and it was reproducing fairly reliable on his setup. I had him test again with master and it went away.

closed 2022-03-15 by mrjerryjohns