project-chip/connectedhomeip #14776: 4 chunked read reports that do not get a status response will completely break reads/subscriptions

bzbarsky-apple, 2022-02-04:
#### Problem
Steps to reproduce:
1. Compile chip-all-clusters-app and chip-tool.
2. Run chip-all-clusters-app.
3. Commisission via `chip-tool pairing onnetwork 17 20202021` (node id 17).
4. Prepare to run `chip-tool any read-by-id 0xFFFFFFFF 0xFFFFFFFF 17 0` under a debugger.
5. Add a breakpoint on `ReadClient::ProcessReportData`.
6. Run under the debugger.  When the breakpoint is hit just wait until the server does 3 CRMP resends and gives up.
7. Repeat step 6 three more times.
8. Try to do a read against the all-clusters-app.  Observe no more `ProcessReportData` calls.

On the _server_ side what happens here is that when we land in `ReadHandler::SendReportData` and `IsPriming()` is true.  So we never call `SetResponseTimeout` on the exchange context that was just allocated to handle the read, and it stays at infinite response timeout time.

Now we send our report and start waiting for the status response (because we are chunking for this wildcard read), which never comes.  Since we have no timeout on the exchange, we sit there forever, never call `ReadHandler::OnResponseTimeout`, never reach `~ReadHandler` (which is a problem on its own), never call `OnReportConfirm` on the reporting engine.  So `mNumReportsInFlight` never gets decremented, and then we hit our cap of 4 of those and stop being able to handle any more.

 #### Proposed Solution
At a bare minimum, we need to do the `SetResponseTimeout` unconditionally in `ReadHandler::SendReportData`.

But that's not enough, because when we time out we call `Close`, which moves us to the `AwaitingDestruction` state.   So when `~ReadHandler` is called `IsAwaitingReportResponse()` (which is a _state check_) is false, and we still don't call `OnReportConfirm`.

@yunhanw-google what is the state-management here supposed to look like?  Should we just be calling `OnReportConfirm` in `ReadHandler::OnResponseTimeout`, if we're `IsAwaitingReportResponse()` there before we `Close()`?

yunhanw-google, 2022-02-21:
fyi, during read interaction, we would not expect status response for non-chunked report

yunhanw-google, 2022-03-03:
https://github.com/project-chip/connectedhomeip/pull/14805

commit 51d1620c (2022-02-18): Fix #14776

--SetResponseTimeout when it is subscribe or in chunked procedure
--Call OnReportConfirm in MoveToState when state is IsAwaitingReportResponsee
closed 2022-03-03 by yunhanw-google